Klipper3d/klipper #1313: install Klipper on macOS

Skorpi08, 2019-02-25:
How can i install Klipper on Mac?
install-octopi.sh dont work with Mac

Maglin, 2019-02-25:
Install it on a Linux machine.  macOS isn't Linux.

Hywelmartin, 2019-02-25:
install-OCTOPI.sh  is only for Octopi
If you know what you are doing it's maybe  possible
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3

KevinOConnor, 2019-03-04:
I'm going to close this as I think the original question has been answered (it's possible but it's not something we can support).

-Kevin

closed 2019-03-04 by KevinOConnor

trevjonez, 2019-03-04:
I spent some time looking at getting klipper to run cleanly on osx ("for science"). 

you can likely get all the correct toolchains installed for the arm and avr firmware using homebrew but I would generally recommend using a docker container to do said compilation work so that it is portable and consistent. using docker to do that sort of thing is possible even when using the interactive `make menuconfig` but is not beginner friendly at all so keep that in mind if anyone goes down that rabbit hole.
That is were docker for mac and klipper must part ways. see this issue as to why you can't run klippy host inside docker for mac.

osx tty docker issue: https://github.com/docker/for-mac/issues/900

then to replace the service bits for klipper on osx you would probably want to write a launchd plist. 

how to launchd job: https://developer.apple.com/library/archive/documentation/MacOSX/Conceptual/BPSystemStartup/Chapters/CreatingLaunchdJobs.html