Klipper3d/klipper #1317: Endstop: Clarification on endstop and stepper collocation requirement

trevjonez, 2019-02-26:
I have been contemplating trying to implement a custom filtering algorithm for virtual end stop signaling as a way to make them more reliable. In doing so it appeared that it would require said filtering to be implemented on the MCU itself as the endstop mcu logic directly communicates with the stepper logic. 

I assume that this direct endstop to stepper mcu level logic is what drives the requirement of endstop and stepper to be on the same MCU? Is there anything more to this that would be a cause for that requirement? is the the latency of MCU^a -> host -> MCU^b large enough to prohibit letting the host manage the interaction logic?

FHeilmann, 2019-02-26:
the 'issue' is that the host software runs somewhat asynchronous to the printer, so one would have to sync up the python code on the host and the two MCUs to achieve a latency that is low enough to work reliably.

Thats why its easier to have the mcu stop the steppers directly and then notify the host of a triggered endstop.

KevinOConnor, 2019-02-26:
It's also necessary to take into account communication failures.  So, if one issues the homing sequence to the micro-controller, and then there is a communication failure, the stepper should still halt at (or prior to) the endstop.

Handling that error situation correctly is non-trivial.  I doubt it would be worth the development effort.

-Kevin

closed 2019-02-27 by trevjonez