Klipper3d/klipper #1319: Feature request (or did I miss this in the docs?): Ability to set flow percentage in printer.cfg

rdeker, 2019-02-26:
I'm using klipper and love it, thanks for the excellent software!

I've noticed that the "Tune" menu has a flow percentage setting, which has saved my butt on a few occasions where I missed setting my properly calibrated flow in my slicer. Is there some way to either save that tuning option, or set a default flow percentage in the printer config so I don't have to contend with checking and re-checking every Cura profile for each print, or is there some reaon that this would be a "bad idea"?

Any help is appreciated.

FHeilmann, 2019-02-27:
I think all the major firmwares start out with 100% after a  reset. If you find yourself constantly adjusting your flow in your slicer, you should calibrate your esteps instead.

Hywelmartin, 2019-02-27:
The flow-% that is set in the slicer isn't something that the printer is aware of..its not in the  gcode..
as @FHeilmann suggested, it seems like you'r having a miss-callibrated extruder

rdeker, 2019-02-28:
Thanks for the input. I had calibrated e-steps based on a straight "extrude 100 mm and measure the offset from reality of filament fed" method. I'll try just applying the necessary percentage offset into the steps/mm value and try that.

closed 2019-02-28 by rdeker