Klipper3d/klipper #1320: Simultaneous usage of mixed TMC SPI & UART configurable drivers

andrico21, 2019-02-27:
Is it possible in Klipper to use SPI-controlled TMC2130 and UART-controlled TMC2208 drivers the same time?
I have 4 x TMC2130 working without any problems on Due board now and just need to add 3 more drivers on the **second Due controller board**. The only drivers I have now are TMC2208, so is it possible to use SPI and UART for different drivers (or boards) the same time?
p.s. for example, XYE steppers - TMC2208, Z1, Z2, Z3, Z4 steppers - TMC2130.

apsbiker, 2019-02-28:
I am actually doing exactly that with a Rumba Plus board. TMC2130 on the X and Y, TMC2208 on the Z and E drivers. As long as you have pins available it absolutely works to mix SPI and UART drivers.

KevinOConnor, 2019-03-04:
I'm going to close this as I think the original question has been answered (Yes, it is possible).

-Kevin

closed 2019-03-04 by KevinOConnor