Klipper3d/klipper #1322: Lost communication with MCU 'mcu'

smokez89, 2019-02-27:
Okay so realise there have been a number of issues raised re. this, but not sure how case specific they are. My knowledge is not the best as relatively new on the klipper scene and have tried fishing through the log to see if anything stands out. Wondering if someone more knowledgeable can spot an issue.

Had a fresh install of klipper running fine after updating my Ender 3s board to a MKS GEN L with TMC2208s. Printed board housing last night 6+ hours for first print since install, worked flawlessly. Had to remove board to fit the enclosure and rewire, powered back on to ensure all was well and all seemed fine. 

Went to start a print this afternoon, and after bed Bagan heating I noticed the LCD had frozen and I received the lost communication with MCU error.

Looking for advice as to wether this is software or hardware related as nothing has changed hardware wise from yesterday, the only difference I can think of is slicer settings on the new file o attempted to print this afternoon but cannot see that having an impact, although I may be wrong. Excuse my ignorance/lack of knowledge, been a real learning curve considering I'm usually a daft hand at technical stuff.

Any help much appreciated. Log attached.

Thank you

[klippy.log](https://github.com/KevinOConnor/klipper/files/2911224/klippy.log)

jakep82, 2019-02-27:
You noted that it worked flawlessly, and then you took it all apart and rewired everything before having this problem.  That would indicate to me that this is likely hardware related.  Check to make sure none of your connections are loose.

smokez89, 2019-02-27:
Agreed. Will be doing so this evening when I get home. Just thought as I had the log on my phone I would upload in case there was anything obvious there as on inspection earlier it appeared all was well. More checks on the hardware are without a doubt needed though. Noted 👍🏼

KevinOConnor, 2019-03-07:
Note the FAQ: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-keep-getting-random-lost-communication-with-mcu-errors

closed 2019-03-22 by KevinOConnor

Spartacus1066, 2019-03-23:
Happened with me again... 3x this week, it's not a USB cable or PS issue unless all 3 I tried are bad that work on other machines. I am 2 hrs 40 minutes into a print... locked up. Crap.
If there was even a way to resume...