Klipper3d/klipper #1324: Touch-mi

le-belge, 2019-02-27:
Hello, I installed Klipper, it works well but I would like to configure a leveling sensor, could you help me? 
https://www.hotends.fr/fr/home/64-capteur-auto-leveling-touch-mi.html
printer : Creality Ender 3

Thank you

I'm French , and I use google translation...

Hywelmartin, 2019-02-28:
It seems like an ordinary end-stop-probe and some macro to start/end the probing should be sufficient  

https://github.com/KevinOConnor/klipper/blob/master/config/sample-probe-as-z-endstop.cfg
https://github.com/KevinOConnor/klipper/blob/5f5d39ef3fae1029cccbbcdf0ba0d009b89c0f09/config/example-extras.cfg#L21
https://github.com/KevinOConnor/klipper/blob/5f5d39ef3fae1029cccbbcdf0ba0d009b89c0f09/config/example-extras.cfg#L1266

le-belge, 2019-02-28:
hello, thanks for the link, I will try

Mike

deudtens163, 2019-03-05:
here are the sections I used to my Touch-MI on my Dagoma Discovery.
Touch-MI probe is automatically deployed when X axis is homed by the magnet.
Retraction of the probe is done at the beginning of the print.

[probe]
pin: P1.25
x_offset: -10
y_offset: -44
z_offset: 4.65
speed: 3.0
#activate_gcode:
#deactivate_gcode:

[homing_override]
set_position_z: 5
axes: z
gcode:
    G92
    G1 Z12 F600
    G90
    G28 X0 Y0
    G1 X100 Y100 F3600
    G28 Z0
    G1 Z5

[stepper_z]
step_pin: P0.20
dir_pin: P0.21
enable_pin: !P0.19
step_distance: .0003125
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0.00
position_min: -5
position_max: 180

le-belge, 2019-03-06:
Hello, thank you very much for the information, I test it quickly !!!

KevinOConnor, 2019-03-22:
I'm going to close this as it appears the discussion has concluded.

-Kevin

closed 2019-03-22 by KevinOConnor

romainbavera, 2019-04-02:
Hi,

I've followed previous process but following error message appears in octoprint session:

19:14:33 File contains parsing errors: /home/pi/printer.cfg
[line 61]: 'G92\n'
[line 62]: 'G1 Z12 F600\n'
[line 63]: 'G90\n'
[line 64]: 'G28 X0 Y0\n'
[line 65]: 'G1 X100 Y100 F3600\n'
[line 66]: 'G28 Z0\n'
[line 67]: 'G1 Z5\n'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready
19:14:33 File contains parsing errors: /home/pi/printer.cfg

[printer.txt](https://github.com/KevinOConnor/klipper/files/3035383/printer.txt)

Thanks for your support.

Regards,
Romain.

jakep82, 2019-04-02:
`
[Touch-MI]
#sensor_pin: ^PC4
#pin_move_time: 0.2
pin: ^PC4
x_offset: 45
y_offse?t: -5
# Increase Z_Offset to lower nozzle closer to bed. PLA 2.6
#Augmenter la valeur z_offset va descendre la buse plus près du bed. Réduire la valeur mettra plus de distance.
z_offset: -2.7
speed: 3.0
`

Touch_MI is not a valid section header, and you have a typo in the y offset line.

romainbavera, 2019-04-02:
Thanks jakep82!
I will follow your tips.

and what is your advice regarding:

[line 61]: 'G92\n'
[line 62]: 'G1 Z12 F600\n'
[line 63]: 'G90\n'
[line 64]: 'G28 X0 Y0\n'
[line 65]: 'G1 X100 Y100 F3600\n'
[line 66]: 'G28 Z0\n'
[line 67]: 'G1 Z5\n'

jakep82, 2019-04-02:
G92 is incorrect.  Your homing override should be changed.

```
[homing_override]
set_position_z: 5
axes: z
gcode:
    G1 Z12 F600
   G28 X0 Y0
   G1 X100 Y100 F3600
   G28 Z0
   G1 Z5
```

polhomarkho, 2019-08-17:
Hi, since I struggled a bit to configure my touch mi with my ender 3, here is my config file if it can helps.
The ender 3 is stock and the touch mi is installed following the hotends.fr instructions. Hope it helps :)

```
## touch mi
[probe]
pin: ^PC4
x_offset: -47  # 47mm left offset from nozzle
y_offset: -11.7  # 11.7mm front offset from nozzle 
speed: 3.0
#activate_gcode:
#deactivate_gcode:

[homing_override]
set_position_z: 5
axes: z
gcode:
   G90
   G1 Z20 F600
   G28 X0 Y0
   G1 X117.5 Y117.5 F3600  # center the nozzle
   G28 Z0

[stepper_z]
step_pin: PB3
dir_pin: PB2
enable_pin: !PA5
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0.0
position_min: -5
position_max: 250

## Create the g29 command for mesh calibration
[gcode_macro G29]
gcode:
   G28
   G1 Z10 F600
   BED_MESH_CALIBRATE

[bed_mesh]
speed: 100
horizontal_move_z: 10
min_point: 60,30
#   Defines the minimum x,y position to probe when for rectangular
#   beds. Note that this refers to the nozzle position, take care that
#   you do not define a point that will move the probe off of the bed.
#   This parameter must be provided for rectangular beds.
max_point: 230,230
#   Defines the maximum x,y position to probe when for rectangular
#   beds. Follow the same precautions as listed in min_point. Also note
#   that this does not necessarily define the last point probed, only
#   the maximum coordinate. This parameter must be provided.
probe_count: 5,5
#   For rectangular beds, this is a comma separate pair of integer
#   values (X,Y) defining the number of points to probe along each axis.
#   A single value is also valid, in which case that value will be applied
#   to both axes.  Default is 3,3.

## from https://3dprintbeginner.com/install-klipper-on-ender-3/

[stepper_x]
step_pin: PD7
dir_pin: !PC5
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC2
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PC6
dir_pin: !PC7
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC3
position_endstop: 0
position_max: 235
homing_speed: 50

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB1
dir_pin: !PB0
enable_pin: !PD6
step_distance: 0.010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
min_temp: 0
max_temp: 260
pressure_advance: 0.6
pressure_advance_lookahead_time=0.075

[heater_bed]
heater_pin: PD4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA6
min_temp: 0
max_temp: 130

[fan]
pin: PB4

[mcu]
serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AK06S84T-if00-port0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5 
max_z_accel: 100
square_corner_velocity: 5.0

[display]
lcd_type: st7920
cs_pin: PA3
sclk_pin: PA1
sid_pin: PC1
encoder_pins: ^PD2, ^PD3
click_pin: ^!PC0

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# points =
#*# 	-0.007500, 0.295000, 0.437500, 0.707500
#*# 	0.500000, 0.537500, 0.570000, 0.677500
#*# 	0.577500, 0.697500, 0.765000, 0.765000
#*# 	0.860000, 1.075000, 1.127500, 1.095000
#*# x_count = 4
#*# y_count = 4
#*# min_x = 0.0
#*# max_x = 174.99
#*# min_y = 10.0
#*# max_y = 224.98
#*# x_offset = 48.0
#*# y_offset = -2.0
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 27.309
#*# pid_ki = 1.701
#*# pid_kd = 109.576
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 68.539
#*# pid_ki = 1.460
#*# pid_kd = 804.472
#*#
#*# [probe]
#*# z_offset = 5.198
```