Klipper3d/klipper #1326: Global name 'homing is not defined in bed_screws.py

JoaoFelipe, 2019-02-28:
Line with error: https://github.com/KevinOConnor/klipper/blob/b40a04fb8d9262419517773434a366f496ca82f8/klippy/extras/bed_screws.py#L55

I got the following exception:
```
Internal error on command:"BED_SCREWS_ADJUST"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 195, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 75, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/bed_screws.py", line 86, in cmd_BED_SCREWS_ADJUST
    self.move((None, None, self.horizontal_move_z), self.speed)
  File "/home/pi/klipper/klippy/extras/bed_screws.py", line 55, in move
    except homing.EndstopError as e:
NameError: global name 'homing' is not defined
```

It seems easy to fix: adding "import homing" should fix it.
I'm still trying to make klipper work and I don't know how to reload the source without restarting the machine, so I won't submit a PR now.

KevinOConnor, 2019-02-28:
Oops - thanks.  Should be fixed now (commit 5f5d39ef).

FYI - to upgrade the software and/or software restarts see: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software

-Kevin

JoaoFelipe, 2019-02-28:
Fixed!

closed 2019-02-28 by JoaoFelipe