Klipper3d/klipper #1329: Beahvior of PROBE_CALIBRATE seems wrong in several ways?

Stephan3, 2019-02-28:
Dear @KevinOConnor.

I spent some time this week with the PROBE_CALIBRATE command on my corexy machine.
If i issue the command the probe triggers the heigth at about 1mm.

The calculated offset value PROBE_CALIBRATE resulting in is -5.12 which does not makes sense to me. Maybe i got the purpose of this command wrong. As i understood the z offset:
If Z0 is the bed surface and probe triggers at 2mm, my probe z offset is 2mm.

Therefor PROBE_CALIBRATE should result in 2mm as a offset.
The current logic retracts the probe by 5mm after probing.(gets called by PROBE_CALIBRATE )
```self.cmd_PROBE(params)
        # Move away from the bed
        toolhead = self.printer.lookup_object('toolhead')
        curpos = toolhead.get_position()
        curpos[2] += 5.```

 and this get into the finalize function of probe.py. wher thing confusing me even more.
´´´z_pos = self.z_offset - kin_pos[2]´´´
So probe triggers(z=2mm), then it retracts by 5mm(Z=7mm), then oldoffset-currentposition=new offset?

Whats the reason it is done that way? What do i miss? Why not simply probe n times at a given postion and the average of this probings is the new probe offset? could be set to self.average? in CMD_PROBE_ACCURACY for ex?

Thank for paying attention

joonadec, 2019-02-28:
> If Z0 is the bed surface and probe triggers at 2mm, my probe z offset is 2mm.

Probe offset is actually the distance of the nozzle to the bed, not the distance of the probe to the bed.

Stephan3, 2019-03-01:
@joonadec  Yes. Sorry if i explained it not clear enough. Z position is allways related to nozzle position, cause this is what the positioning is all about.

KevinOConnor, 2019-03-02:
It's entirely possible something is wrong with the new code.  However, without a log from the event, there's little we can do.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Stephan3, 2019-03-03:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2923005/klippy.log)

IF a PR is welcome, i allread have a solution for that.

KevinOConnor, 2019-03-04:
Hi Stephan.  I don't understand what is being requested or raised here.  If you're reporting that some probes are sensitive to temperature then I don't think that is an issue with the current code as that is not the intent of the PROBE_CALIBRATE command.  The documentation for PROBE_CALIBRATE does specify to run at room temperature - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Level.md .  I don't see how one could get a 5mm offset from temperature, however.

Otherwise, if you've found a defect in the code then, yes, please describe the code defect or open a pull request with the fix.

If it's not temperature and/or you are unsure if there is a defect, then you'll need to reproduce on the pristine Klipper code, provide the klipper log file from that, and I'll take a look.  If so, please grab the latest code as I've added some additional logging.

Cheers,
-Kevin

Stephan3, 2019-03-08:
Hi @KevinOConnor.

I reconstructed the Workflow now several times. And yes, now i am sure that this is a issue, for those who arent homing with that probe. Let me try to explain the workflow the current codebase looks like to me.

User is homing with mechanical endstop and issues PROBE_CALIBRATE. We are entering now probe.py method
```  python  
def cmd_PROBE_CALIBRATE(self, params):
        # Perform initial probe
        self.cmd_PROBE(params)
        # Move away from the bed
        toolhead = self.printer.lookup_object('toolhead')
        curpos = toolhead.get_position()
        curpos[2] += 5.
        toolhead.move(curpos, self.speed)
        # Move the nozzle over the probe point
        curpos[0] += self.x_offset
        curpos[1] += self.y_offset
        toolhead.move(curpos, self.speed)
        # Start manual probe
        manual_probe.ManualProbeHelper(self.printer, params,
                                       self.probe_calibrate_finalize)
```
It probes the bed wit `self.cmd_PROBE(params)` then it moves the toolhead 5mm up `curpos[2] +=5.`
Okay so lets asume the probe triggers at heigth of 1.0 mm. I am talking about the height, the nozzle has at the moment of triggering. Now the next step enters the ManualProbeHelper method:

``` python
class ManualProbeHelper:
    def __init__(self, printer, params, finalize_callback):
        self.printer = printer
        self.finalize_callback = finalize_callback
        self.gcode = self.printer.lookup_object('gcode')
        self.toolhead = self.printer.lookup_object('toolhead')
        self.speed = self.gcode.get_float("SPEED", params, 5.)
        self.past_positions = []
        self.last_toolhead_pos = self.last_kinematics_pos = None
        # Register commands
        try:
            self.gcode.register_command('ACCEPT', self.cmd_ACCEPT,
                                        desc=self.cmd_ACCEPT_help)
        except self.printer.config_error as e:
            self.gcode.respond_error(
                "Already in a manual Z probe. Use ABORT to abort it.")
            self.finalize_callback(None)
            return
        self.gcode.register_command('NEXT', self.cmd_ACCEPT)
        self.gcode.register_command('ABORT', self.cmd_ABORT,
                                    desc=self.cmd_ABORT_help)
        self.gcode.register_command('TESTZ', self.cmd_TESTZ,
                                    desc=self.cmd_TESTZ_help)
        self.gcode.respond_info(
            "Starting manual Z probe. Use TESTZ to adjust position.\n"
            "Finish with ACCEPT or ABORT command.")
        self.report_z_status()
    def get_kinematics_pos(self):
        toolhead_pos = self.toolhead.get_position()
        if toolhead_pos == self.last_toolhead_pos:
            return self.last_kinematics_pos
        self.toolhead.get_last_move_time()
        kin_pos = self.toolhead.get_kinematics().calc_position()
        self.last_toolhead_pos = toolhead_pos
        self.last_kinematics_pos = kin_pos
        return kin_pos
    def move_z(self, z_pos):
        curpos = self.toolhead.get_position()
        try:
            if curpos[2] - z_pos < Z_BOB_MINIMUM:
                curpos[2] = z_pos + Z_BOB_MINIMUM
                self.toolhead.move(curpos, self.speed)
            curpos[2] = z_pos
            self.toolhead.move(curpos, self.speed)
[... 86 lines cut ...]
Now `kin_pos[2]` is still 6. mm. Next the code calculates the new offset. I asume here old offset is 0mm.
`z_pos = self.z_offset - kin_pos[2]`
==
`new_offset = old_offset - current_z_pos`
==
`-5. = 0. - 6.`

This does not make sense to me, as is.

So the only way the current code can work if a user drives the probe manualy to the bed using cmd_TESTZ 0.0.
Then the math would be:
`z_pos = self.z_offset - kin_pos[2]`
==
`new_offset = old_offset - current_z_pos`
==
`1.2 = 0. - -1.2.`

So this workflow only has sense(as i understood it) if you want to home the printers Z with your probe.
It does not cover the boys using the probe only for z_tilt_adjust, quad_gantry_level, bed_mesh etc.

Therefor i would propose a method which can set a new offset during runtime for those who want to deal whith tempdrift of the probe. So one can set the new offset automaticaly b4 every z_tilt_adjust, quad_gantry_level, bed_mesh etc.

What do you think?

- Stephan

KevinOConnor, 2019-03-08:
Thanks.  The PROBE_CALIBRATE command was indeed not working correctly.  I think it should be fixed now (commit 1b1d3f15).  Alas, my printer with the automatic probe is in parts right now, so I've only been able to run unit tests on this.

I'm not sure what you are proposing with regard to temperature.  The general approach I'd recommend is to perform bed levelling at room temperature, and then use a static offset to account for differences introduced by temperatures using a SET_GCODE_OFFSET command.  This is described at https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Level.md#determining-thermal-expansion .

I'm also aware that @Arksine has done some work on automatically applying an adjustment to induction sensors via a thermistor.

-Kevin

Stephan3, 2019-03-08:
@KevinOConnor 

I had tested this commit on two pinters now. The result is correct. Movement of small floatingpoint numbers is strange.

User did:
`PROBE_CALIBRATE`
then klipper probed z _height and stored into self.probe_calibrate_z 
Thats okay so far. 
Now user has to adjust zheigth with TESTZ. You ever tried that :D ?
It doesent handle floating point numbers, smaller than .4, right? You can `TESTZ Z=-1` or `TESTZ Z=-5` but not `TESTZ Z=.1` because z values like Z=.25 or Z=-.2 result in a zhop like movement. it moves down and imediatly back up. the endpostion is right then. Is this zhop ecxpected?

---

For the proposal i made. There can be a nother method, lets say PROBE_ADJUST_OFFSET.
It will probe uservalue times at Xuservalue Yuservalue position. The mean of theese probings are new offset. It is directly set to self.z_offset in probe.py.

Why is this useful? Everytime you do a mesh or a gantry level a perfect probe offset is necessary.  Your offset is´nt allways the same as a probe heatsoaks. I ecxperienced deviations for capacitive probes of .3, bltouch .4, Fotek pl08 .3 and piezo much more. Theese values arent guessings, they are messured with a clean messurement plan. And no it is not a linear behavior. Maybe @Arksine had come to that conclusion also.

So in each condition of the printer, one can set perfect offset during runtime using PROBE_ADJUST_OFFSET. I use this method now since 3 days before quad_gantry_level and it does well, needs no manual touching or anything. But here again this works only for those not homing with that probe. Therefore i dont think that SAVE_CONFIG is neccessary with this one.

Thanks 

-Stephan

KevinOConnor, 2019-03-11:
I'm not really sure what you are reporting with the TESTZ command.  The command will move a relative distance on each invocation.  If the target Z position is within .5mm of the current Z position, then the TESTZ command will first raise the nozzle (relative to the bed) and then descend to the target position.  This is intentional and a requirement - otherwise small moves would not occur correctly due to static friction.  If you're seeing an issue with TESTZ then please run the command that does something unexpected, issue an M112, and attach the Klipper log file.  That allows everyone to see exactly what was requested and exactly what actions the printer took.

As for calibrating the probe offset from a known endstop position - that's fine if someone wishes to add support for it.

-Kevin

KevinOConnor, 2019-03-22:
I'm going to close this as I think the original bug is now fixed.

-Kevin

closed 2019-03-22 by KevinOConnor