Klipper3d/klipper #1330: TMC2208 - stepper motors hotter on UART

kitamo1981, 2019-02-28:
Hi,

I have a MKS GEN L and TMC2208 record. I was all very well aware, however, I noticed that extruder sometimes suspends its work and stops working during printing.
I decided that I will try to control it using UART to select more detailed work parameters. The connection was successful, everything works as it should - printouts do not suspend the extruder, but the stepper motor on the extrude player is now much hotter than before.

I am trying to reduce the voltage in the configuration file but it seems to me that it does not change anything. On the potentiometer I have a voltage like the previous one, which is 0.9V, which ensured very good work.

any suggestion what can cause such an effect? on other engines the temperature also increased after connecting UART.

>[extruder]
step_pin: PA4
dir_pin: !PA6
enable_pin: !PA2
step_distance: 0.010328
nozzle_diameter: 0.400
filament_diameter: 1.750
pressure_advance: 0.250
pressure_advance_lookahead_time: 0.010
heater_pin: PB4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PK5
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
min_temp: 0
max_temp: 260
max_extrude_only_distance: 200

>[tmc2208 extruder]
uart_pin: ar44
microsteps: 16
#interpolate: True
run_current: .9
hold_current: .5
stealthchop_threshold: 500

jakep82, 2019-02-28:
The run_current setting is current (amps), not volts.  I believe a vref of 0.9 volts on a 2208 would be 0.7 amps, so try reducing your run_current to 0.7.

kitamo1981, 2019-02-28:
ou, *1.3 so I just got wrong calculations :D I will try.
What kind of hold_current I should set? how to calculate it?

jakep82, 2019-02-28:
Your current setting is fine for hold_current.

KevinOConnor, 2019-03-11:
I'm going to close this as it looks like the original question was answered (reduce run_current to reduce heat).

-Kevin

closed 2019-03-11 by KevinOConnor