Klipper3d/klipper #1332: Hard or emergency end stops for machine limits when using a BLTouch as a virtual endstop

davidjericho, 2019-02-28:
I tried to find an answer, but it's escaping me. 

I use a BLTouch as a virtual endstop, and two MCUs for a 4 Z screw D-Bot. It normally works very well.

On more than one occasion, the touch sensor hasn't fully deployed as it's over the bed at the time without enough height, and as such the Z homing script is now out of whack, resulting in the bed being pushed up into the nozzles. The springs on the bed absorb the Z motion, but now I have a large amount of force on the printer carriage and other bits until I force a move. 

My existing homing_override reads:
`[homing_override]
axes: z
gcode:
    G90
    G0 Z5 F18000 ; Move bed 5mm away
    G28 X0 Y0 ; Home X and Y
    G1 X150 Y135 F18000 ; Move to centre of bed
    G28 Z0 ; Home Z
set_position_z: 0.0
`

In Smoothieware, I was able to set the alpha_limit_enable (and beta, gamma and so on) end stop options, that allow me to hard stop the machine should I exceed a limit mechanically I don't want to go past. Given the speed Klipper lets me run this printer at, and the power of the steppers, I'd really like to be able to break an endstop and stop the machine, rather than slam an expensive print head into something :) 

Would it be possible to implement a hard end stop function such that it limits or stops movement once touched, regardless of what the virtual end stop permits? I think an additional benefit would be automatic bed sizing with end stops at all 6 extremes of movement.

jakep82, 2019-02-28:
You could add a regular endstop to your Z axis and use the bltouch as just a probe.  That said, when the bltouch doesn't fully deploy, it should enter an error state that prevents the printer from moving.

davidjericho, 2019-02-28:
I am using the virtual end stop because my bed is removable, and I change materials and coverings of my glass depending on the material I'm printing. It's an easy way to adjust the 4 z screws (when used with z_tilt_adjust) for accurate alignment.

KevinOConnor, 2019-03-04:
In order for us to assist, you'll need to attach the Klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

davidjericho, 2019-03-05:
Unfortunately I don't have a log on hand, because each time it has occurred I've found absolutely nothing in the logs to determine why it happened. Right after killing the power, I've rerun the same task and it worked perfectly.

I'll close the issue and re-raise it if it happens again with a full log dump.

As a feature request of additional hard end stops, I'd be keen to know how to ask for this.

closed 2019-03-05 by davidjericho