Klipper3d/klipper #1333: Confirming Behavior of quad_gantry_level with multiple extruders

ArdentSquirrel, 2019-02-28:
I have noticed that quad_gantry_level requires me to run through the quad_gantry_level process several times before the z motors are adequately level to the bed, with each iteration requiring less adjustment to the gantry.  It feels as if the probe offset is causing the slight miscalculation. (The probe is between both extruders, and about 35mm in front of the nozzles)

I have carefully measured the center of z and z2 motors in respect to X=0,Y=0, measuring the center of each motor to the center of the nozzle of the primary extruder.  I have carefully measured the bltouch probe offset as well.  Should I be measuring the z and z2 motors in reference to the nozzle, or to the BLTouch probe, or to middle point between the nozzles?

ArdentSquirrel, 2019-02-28:
[klippy_20190228.log](https://github.com/KevinOConnor/klipper/files/2917001/klippy_20190228.log)

Uploaded latest klippy.log

KevinOConnor, 2019-03-04:
FYI, there is a PR for quad_gantry_level - see #1328.  Beyond that, I do not know the answer to your question.

-Kevin

ArdentSquirrel, 2019-03-04:
Thank you @KevinOConnor.  I will pull the updated file and see if it makes a difference.  @mzbotreprap might you know if the offset use the probe location, or the location of the nozzle in reference to where the Z stepper is located?

VoronDesign, 2019-03-04:
Probe offset is taken from [probe] section. It assumes you've already done the calculations needed to position the probe points relative to where your probe is actually located. Once probing is done it will shift the X and Y according to the offsets so the rest of the calculations make sense.

For example, if your probe is 25mm behind the nozzle, and you want to probe a point 25,25 the actual point coordinates you'll use in config is 25,0 and quad_gantry code will assume 25,25 as the probed point coordinate given the correct offset value in [probe]

ArdentSquirrel, 2019-03-04:
Thank you for the clarification @mzbotreprap.  I am noticing that I have to run through quad_gantry 2-3 times before the amount of correction is less than 1mm, with each iteration correcting less and less.  I have updated the code with your proposed pull, and it still seems to take a couple of iterations for it to be truly level.  I was thinking that this was caused by referencing the wrong point in relation to the stepper motor, but I have been using the extruder0 nozzle for my reference.  I'll step through the code and figure out what is going on with the math.  I assume on your machine you only need to run through the level procedure one time?

VoronDesign, 2019-03-04:
There's a known issue with the code if you're using top of trunk. I have a pending PR to fix the logic so you don't have to probe billion times. I'm about to get to +/- 0.04mm on the first round on an especially tweaked gantry.

Keep in mind the motor numbering and order of proving point is also important. The sample config "should" have the Z motor positions outlined in it.

ArdentSquirrel, 2019-03-04:
I just thought of something...I am not using the probe section, and instead am using the bltouch section.  Would quad_gantry still use the bltouch section?

[bltouch]
sensor_pin = z:ar10
control_pin = z:ar11
pin_move_time = 0.675
pin_up_reports_not_triggered = True
pin_up_touch_mode_reports_triggered = True
x_offset = 16.5
y_offset = -36.5
speed = 5.0
z_offset = 1.002

VoronDesign, 2019-03-04:
It should, so long as bltouch is using the standard probe_finalize callback and returns the correct offsets in that callback. So your probe is 16mm to the right and 36mm in front of the nozzle? What printer are you using? (genuinely curious).

ArdentSquirrel, 2019-03-04:
I am using the Piper2 CoreXY Printer.  And yep it's 16mm to the right and 36 mm in front of the Extruder 0 nozzle.  The probe is directly between both nozzles.

[image]

ArdentSquirrel, 2019-03-05:
@mzbotreprap Reading through your proposed code change, I see this line:

self.gcode.respond_info("Actuator Positions:\n z1: %0.6f\n z2: %0.6f\
            \n z3: %0.6f\n z4: %0.6f\n" % (
                z_height[0],z_height[1],z_height[2],z_height[3]))

Should this not be:

self.gcode.respond_info("Actuator Positions:\n z0: %0.6f\n z1: %0.6f\
            \n z2: %0.6f\n z3: %0.6f\n" % (
                z_height[0],z_height[1],z_height[2],z_height[3]))

...just to keep it consistent with the gantry relative probe points?

I do like the output:

18:52:07 probe at 10.000,30.000 is z=0.879500
probe at 10.000,30.000 is z=0.879500
probe at 10.000,300.000 is z=0.889500
probe at 10.000,300.000 is z=0.894500
probe at 275.000,300.000 is z=0.884500
probe at 275.000,300.000 is z=0.879500
probe at 275.000,30.000 is z=0.902000
probe at 275.000,30.000 is z=0.904500
Gantry-relative probe points:
z0 = 4.120500
z1 = 4.108000
z2 = 4.118000
z3 = 4.096750
Actuator Positions:
z1: 4.132116
z2: 4.101422
z3: 4.124533
z4: 4.086232
Average: 4.111076
Making the following gantry adjustments:
stepper_z = -0.021040
stepper_z1 = 0.009654
stepper_z2 = -0.013457
stepper_z3 = 0.024844

VoronDesign, 2019-03-05:
Oh that’s more like it :) And good suggestion. No reason to mix labels 

I’ll update the PR

KevinOConnor, 2019-03-22:
Was there a solution found for this?

-Kevin

ArdentSquirrel, 2019-03-22:
FR https://github.com/KevinOConnor/klipper/pull/1328 Made this much better.  Additionally, to answer the question, the quad_gantry_level process measures to the Extruder0 nozzle.  Closing this issue.

closed 2019-03-22 by ArdentSquirrel