Klipper3d/klipper #1334: Restricted max movement on 1mm and .1mm movements (menu)

JoaoFelipe, 2019-02-28:
The `input_max` configuration on 10mm is different from 1mm and .1mm. Since this value defines the actual position in the bed, it is strange to use different `input_max` values. I think it would be better to read the `position_max` configurations from `printer.cfg` (is it possible?). 

I found this bug when I tried to move the head to the position 115 x 115. I managed to move x to 120 using the 10mm movement option. But when I tried to subtract 5mm using the 1mm option, it went from 120mm to 100mm in a single step.

10mm config for x:
https://github.com/KevinOConnor/klipper/blob/6ccc0732c119e9f92c714c4542d760ea77bcbfdd/klippy/extras/display/menu.cfg#L222-L229

1mm config for x:
https://github.com/KevinOConnor/klipper/blob/6ccc0732c119e9f92c714c4542d760ea77bcbfdd/klippy/extras/display/menu.cfg#L270-L277

0.1mm config for x:
https://github.com/KevinOConnor/klipper/blob/6ccc0732c119e9f92c714c4542d760ea77bcbfdd/klippy/extras/display/menu.cfg#L318-L325

It is probably necessary to change x, y, and z.

If it is not possible to read the `position_max` from `printer.cfg`, it would be desirable to at least change everything to 200 (or maybe bigger to support more printers).

bruce356, 2019-02-28:
@JoaoFelipe Hi, I do not see that there is a problem (maybe I am wrong) you can change those numbers to anything that you like. If you do not want to change the default "menu.cfg" file then add your changes/changed entries to your "printer.cfg" file.
Regards - bruce

JoaoFelipe, 2019-02-28:
The problem is just the inconsistency. Why do the input_max parameters have different values in the default config when they have the same meaning?

amblidex, 2019-03-01:
I've run into this problem myself. It does seem strange that the default max values are different for different step values. It would be nice if you could pull the axis ranges from config, but I don't think that is currently supported. At a minimum though, I agree they should all be the same value.

KevinOConnor, 2019-03-07:
It's fine if someone wishes to submit a pull request to change the default menu.  There is work in progress to make the menu and gcode_macros more extensible (see #1141), however it's likely a few months out.

-Kevin

KevinOConnor, 2019-03-09:
FYI, this should be addressed as of commit e2af1c1f2.

-Kevin

closed 2019-03-22 by KevinOConnor