Klipper3d/klipper #1336: temperature sensor.py

JamieAKing, 2019-03-01:
me again 
I'm not usually this inept but I have clearly missed a step somewhere but I cannot find it , something about defining a heater. example config makes no reference to it thou

help appreciated 

[klippy.log](https://github.com/KevinOConnor/klipper/files/2920844/klippy.log)

FHeilmann, 2019-03-02:
seems like your code has gotten out of sync somewhere. One part of the python code is trying to call upon a function that doesn't exist anymore.

Try to make sure that your klipper directory is not screwed up in any way (worst case, re-clone the repository) and see if that helps.

JamieAKing, 2019-03-02:
im afraid this Is where my limited knowledge runs out. I can make just about anything ou tin the garage  but the computer side of this project is definitely a weak point. is there a tutorial out here somewhere on how to keep upto date with the current git repo. obviously I'm not doing so. I 'cloned?' (downloaded) the master code file and copied the new temp sensor.py in extras to the pi via ssh. (get me) are there other files that I should copy?

JamieAKing, 2019-03-02:
right I copied the entire slippy master folder freshly downloaded from this repo to the klipper folder on the host PI3 still the same internal error

closed 2019-03-03 by JamieAKing