Klipper3d/klipper #1337: PT100 with amp board and 4.7K resistor (Prusa MK3 Einsy)

ODINITE, 2019-03-01:
I'm trying to configure my Prusa MK3 and am having an issue with the temperature reading of the PT100. I'm using a Prusa Mk3 with a PT100 amplifier board.

In Marlin, I believe this setting was 1047 if that helps.

Any help is appreciated!

ODINITE

jakep82, 2019-03-01:
Please include a log.

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#issue-reporting

ODINITE, 2019-03-01:
Here is my configuration and the klippy log

I have just done a fresh install and went up to temperature check.

The bed reports 22C (accurate for the room temperature I'm in)

The nozzle temp is reporting about 40C (I assume because of the 4.7K resistor on the prusa einsy board)

[klippy.log](https://github.com/KevinOConnor/klipper/files/2921181/klippy.log)

[klippy.log](https://github.com/KevinOConnor/klipper/files/2921182/klippy.log)

jakep82, 2019-03-01:
You likely need to adjust the ADC voltage.  The pullup resistor is only a valid configuration for thermistors, and 4.7K is the default value anyways.

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L165

ODINITE, 2019-03-01:
Thanks for your prompt response @jakep82 .

My hardware is set up as a PT100 into an E3D amp board into the default pin on the prusa mk3 einsy rambo.

I assumed the 4.7K pull-up is on the board (I have had trouble finding a proper schematic).

The board uses 5V logic and the PT100 E3D board is being powered off of a 5V pin from the einsy

jakep82, 2019-03-01:
Yes, but I'm sure it's not getting exactly 5V.  Check the input to the amplifier with a multimeter.  The temperature Klipper reports is based on the voltage the amplifier outputs, and that is affected by the input voltage.

ODINITE, 2019-03-01:
now i get it.. will check that right now...
5.02 V

So I should add a line to [extruder]
adc_voltage: 5.02

and restart?

jakep82, 2019-03-01:
Re-reading, yes you need to plug into an analog input that doesn't have a pullup on it.  You will need to refer to the pin diagram of the einsy to find a spare analog pin.

https://github.com/ultimachine/Einsy-Rambo/blob/1.1a/board/Project%20Outputs/Schematic%20Prints_Einsy%20Rambo_1.1a.PDF

jakep82, 2019-03-01:
Pin PK0 which is pin 10 on header P3 is analog, and appears to be available based on your config.

ODINITE, 2019-03-01:
SUCCESS!!!!

Thank you again @jakep82 

FYI Here is another nice pinout that I found very helpful for anyone else that has this issue:

https://reprap.org/mediawiki/images/6/6d/EinsyRambo1.1-connectors-j19-p1-p2-p3.png

closed 2019-03-01 by ODINITE

wtyler3, 2019-05-26:
I know this is a close d issue but I am also trying to implement this and the Pin 10 indicated as the solution for the problem is used by the IR sensor. Can pin 6 be used instead?