Klipper3d/klipper #1339: Why "Error: Section 'bed_mesh' is not a valid config section"?

rmcbc, 2019-03-02:
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-291-g309a47c-dirty'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Sat Mar  2 08:21:50 2019 (1551514910.2 663823.0)
===== Config file =====
[mcu]
serial = /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00
restart_method = rpi_usb

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 1000
max_z_velocity = 8
max_z_accel = 100

[stepper_x]
step_pin = P0.4
dir_pin = P0.5
enable_pin = !P4.28
step_distance = .0025
endstop_pin = P1.29
position_endstop = -14
position_min = -20
position_max = 238
homing_speed = 50

[stepper_y]
step_pin = P2.1
dir_pin = P2.2
enable_pin = !P2.0
step_distance = .0025
endstop_pin = P1.27
position_endstop = -10
position_min = -10
position_max = 220
homing_speed = 50

[stepper_z]
step_pin = P0.20
dir_pin = !P0.21
enable_pin = !P0.19
step_distance = .0025
endstop_pin = probe:z_virtual_endstop
position_min = -10
position_max = 230
homing_speed = 8.0

[extruder]
step_pin = P0.11
dir_pin = !P2.13
enable_pin = !P2.12
step_distance = 0.0012567
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = P2.7
sensor_type = EPCOS 100K B57560G104F
sensor_pin = P0.24
min_temp = 0
max_temp = 250
min_extrude_temp = 170
pressure_advance = 0
max_extrude_only_velocity = 50
max_extrude_only_accel = 5000
max_extrude_cross_section = 0.64
max_extrude_only_distance = 400
control = pid
pid_kp = 25.863
pid_ki = 1.214
pid_kd = 137.720

[heater_bed]
heater_pin = P2.5
sensor_type = EPCOS 100K B57560G104F
sensor_pin = P0.23
min_temp = 0
max_temp = 130
control = pid
pid_kp = 71.960
pid_ki = 2.650
pid_kd = 488.428

[fan]
pin = P2.3

[heater_fan my_nozzle_fan]
pin = P2.4
heater = extruder
heater_temp = 50.0
fan_speed = 1.0

[bltouch]
sensor_pin = P1.25
control_pin = P1.23
x_offset = 35
y_offset = 0
z_offset = 2.55
speed = 3.0
pin_up_reports_not_triggered = False
pin_up_touch_mode_reports_triggered = False
pin_move_time = 0.300

[homing_override]
set_position_z = 0
axes = z
gcode =
        G90
        G1 Z10 F800
        G28 X0 Y0
        G1 X75 Y110 F6000     ;X=110-x_offset
        G28 Z0
        G1 Z15

[bed_mesh]
min_point = -5,30
max_point = 155,190
probe_count = 5,5
horizontal_move_z = 6

[bed_screws]
screw1 = -5,30
screw1_name = front left screw
screw2 = 155,30
screw2_name = front right screw
screw3 = 155,190
screw3_name = rear right screw
screw4 = -5,190
screw4_name = read left screw
horizontal_move_z = 10
speed = 50
probe_speed = 3
screw_thread = CW-M3

[display]
lcd_type = st7920
cs_pin = P0.16
sclk_pin = P0.15
sid_pin = P0.18
encoder_pins = ^P3.25, ^P3.26
click_pin = ^!P2.11
kill_pin = ^!P1.31

[gcode_macro G29]
gcode =
        BED_MESH_CALIBRATE

[gcode_macro G30]
gcode =
        G0 Z+10
        PROBE

[gcode_macro M48]
gcode =
        G1 Z5
        PROBE
        G1 Z5
        PROBE
        G1 Z5
        PROBE
        G1 Z5
        PROBE
        G1 Z5
        PROBE
        G1 Z20

[gcode_macro START_GCODE]
gcode =
        M117 Auto Level
        G28
        BED_MESH_CALIBRATE
        M117 Clean nozzle
        G0 X226 Y99 Z10 F7500
        G0 Z1.5
        G0 X233 Y63 F5000
        G0 X233 Y99 F5000
        G0 X226 Y63 F5000
        G0 X226 Y99 F5000
        G0 X233 Y92 F5000
        G0 X226 Y92 F5000
        G0 X233 Y77 F5000
        G0 X226 Y77 F5000
        G0 X233 Y63 F5000
        G0 X226 Y63 F5000
        G0 X226 Y77 F5000
        G0 X233 Y77 F5000
        G0 X226 Y85 F5000
        G0 X233 Y85 F5000
        G0 X226 Y92 F5000
        G0 X233 Y92 F5000
        G0 X226 Y99 F5000
        G0 X233 Y99 F5000
        G0 Y110 X110 Z10 F5000
        M117 Prime nozzle
        G1 X50 Y6 Z0.2 F6000
        G92 E0
        G1 X155 E20 F400
        G92 E0
        M117 Printing

[gcode_macro END_GCODE]
gcode =
        M104 S0
        M140 S0
        G91
        G1 E-3 F5000
        G1 Z+10
        G90
        G1 Y210 F6000
        M84
        M107
=======================
Extruder max_extrude_ratio=0.266081
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 133, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 130, in _read_config
    pconfig.check_unused_options(config)
  File "/home/pi/klipper/klippy/configfile.py", line 193, in check_unused_options
    section,))
Error: Section 'bed_mesh' is not a valid config section

rmcbc, 2019-03-02:
Solved.

closed 2019-03-02 by rmcbc

urobasa, 2021-01-29:
ссука сложно было написать как?