Klipper3d/klipper #1340: LPC176x (Smoothieboard) not working with latest code

rmcbc, 2019-03-02:
After pull the latests code and compiling the firmware with this options:

[ ] Enable extra low-level configuration options
Micro-controller Architecture (LPC176x (Smoothieboard))  --->
Processor model (lpc1768 (100 Mhz))  --->
[*] Target board uses Smoothieware bootloader
[*] Use USB for communication (instead of serial)
USB ids  --->

Don't get the usb device "/dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00"

I did:
make clean
make menuconfig
make
copy out/klipper.bin to sdcard as firmware.bin. Before the latest code it works perfectly now can't get it to work.

rmcbc, 2019-03-02:
My board is a bigtreetech skr v1.1 and it worked perfectly till now with latest code.

rmcbc, 2019-03-02:
#
#Automatically generated file; DO NOT EDIT.
#Klipper Firmware Configuration
#
#CONFIG_LOW_LEVEL_OPTIONS is not set
#CONFIG_MACH_AVR is not set
#CONFIG_MACH_ATSAM is not set
#CONFIG_MACH_ATSAMD is not set
CONFIG_MACH_LPC176X=y
#CONFIG_MACH_STM32F1 is not set
#CONFIG_MACH_PRU is not set
#CONFIG_MACH_LINUX is not set
#CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="lpc176x"
CONFIG_CLOCK_FREQ=100000000
CONFIG_USBSERIAL=y
CONFIG_FLASH_START=0x4000
CONFIG_LPC_SELECT=y
CONFIG_MACH_LPC1768=y
#CONFIG_MACH_LPC1769 is not set
CONFIG_SMOOTHIEWARE_BOOTLOADER=y

#
#USB ids
#
CONFIG_USB_VENDOR_ID=0x2341
CONFIG_USB_DEVICE_ID=0xabcd
CONFIG_USB_SERIAL_NUMBER="12345"
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
CONFIG_HAVE_GPIO_I2C=y
#CONFIG_HAVE_GPIO_HARD_PWM is not set
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_INLINE_STEPPER_HACK=y

rmcbc, 2019-03-02:
The klipper.bin file is only 964 bytes I think it was around 20Kb.

rmcbc, 2019-03-02:
@KevinOConnor 
Your change in Makefile did this.
Original:
$(Q)$(CC) $^ $(CFLAGS_klipper.elf) -o $@
Changed to:
$(Q)$(CC) $(patsubst %.c, $(OUT)src/%.o,$(src-y)) $(OUT)compile_time_request.o $(CFLAGS_klipper.elf) -o $@

KevinOConnor, 2019-03-02:
Thanks.  Should be fixed now (commit 9430fea2).

-Kevin

wiwatto, 2019-03-04:
my get "klipper.bin" file is only 19.5 kb,  also try to compiling another Micro-controller -- not more than 20 kb)

Micro-controller Architecture (LPC176x (Smoothieboard)) --->
Processor model (lpc1769 (120 Mhz)) --->
[x] Target board uses Smoothieware bootloader
[x] Use USB for communication (instead of serial)
USB ids --->12345
MKS Sgen 32Bit

KevinOConnor, 2019-03-11:
I'm going to close this as this issue should now be fixed.  (FYI, it is normal for Klipper firmware to be under 20KB.)

-Kevin

closed 2019-03-11 by KevinOConnor