Klipper3d/klipper #1342: STM32F103 compile problem

istyszy, 2019-03-02:
Hello, 

I think i have a problem with compiling klipper for STM32F103 (bluePill)
This is the output on octoprint and the result is a 1Kb  bin file.

 Creating symbolic link out/board
  Preprocessing out/stm32f1.ld
  Build Kconfig config file
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/stm32f1/main.o
  Compiling out/src/stm32f1/gpio.o
  Compiling out/src/stm32f1/adc.o
  Compiling out/src/stm32f1/spi.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_tim.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_adc.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_spi.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_gpio.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_exti.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_sdmmc.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_fsmc.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_pwr.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_dma.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_rtc.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_utils.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_i2c.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_rcc.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_dac.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_crc.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_usart.o
  Compiling out/src/../lib/hal-stm32f1/source/stm32f1xx_ll_usb.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/../lib/cmsis-stm32f1/source/system_stm32f1xx.o
  Compiling out/src/stm32f1/serial.o
  Compiling out/src/generic/serial_irq.o
  Building out/compile_time_request.o
Version: v0.7.0-291-g309a47c-20190302_165522-octopi
  Assembling out/src/../lib/cmsis-stm32f1/source/startup_stm32f103xb.o
  Linking out/klipper.elf
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: cannot find entry symbol Reset_Handler; defaulting to 08002000
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: cannot find entry symbol Reset_Handler; defaulting to 08002000
  Creating hex file out/klipper.bin

KevinOConnor, 2019-03-02:
This looks like a dup of issue #1340 - it should be fixed now (commit 9430fea2).

-Kevin

istyszy, 2019-03-02:
Thanks it compiles, 
now to connect to it ?  ( i left enabled : use USB for communication)
serial = /dev/ttyACM0 ? 
___________________
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x70495c70, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.7.0-292-g9430fea FIRMWARE_NAME:Klipper
Send: M21
Recv: ok
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
_____________________
config:
[stepper_x]
step_pin: PA2
dir_pin: PA3
enable_pin: !PA4
step_distance: .0125
endstop_pin: ^PB10
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_y]
step_pin: PA5
dir_pin: PA6
enable_pin: !PA7
step_distance: .0125
endstop_pin: ^PB9
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_z]
step_pin: PA8
dir_pin: PA11
enable_pin: !PA12
step_distance: .0025
endstop_pin: ^PB8
position_endstop: 0.5
position_max: 200

[extruder]
step_pin: PB2
dir_pin: PB1
enable_pin: !PB3
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PA15
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA1
control: watermark
min_temp: 0
max_temp: 130

[mcu]
serial = /dev/ttyACM0
pin_map = arduino
restart_method = arduino

[printer]
kinematics = corexy
max_velocity = 300
max_accel = 3000
max_z_velocity = 30
max_z_accel = 300
motor_off_time = 600
require_home_after_motor_off = false
allow_move_without_home = true

KevinOConnor, 2019-03-04:
If you're experiencing some other issue, you'll need to attach the Klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

istyszy, 2019-03-09:
So, i had some work an did not get to work on my printer, sorry for the delay.
Here is my output from terminal to list SERIAL ports:

pi@octopi:~/klipper $ ls -l /dev/serial/by-id/
total 0
lrwxrwxrwx 1 root root 13 Mar  9 10:39 usb-Klipper_Klipper_firmware_12345-if00 -> ../../ttyACM0

and i also uploded the Log file and the config file.
Thanks for the help.

[printer_cfg.txt](https://github.com/KevinOConnor/klipper/files/2948295/printer_cfg.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/2948296/klippy.log)

KevinOConnor, 2019-03-09:
You have a spurious "config:" as the first line of your config file.

-Kevin

istyszy, 2019-03-09:
i managed to get it communicating.
There were other errors, but now it is ok.

Thanks for all your help!

closed 2019-03-09 by istyszy