Klipper3d/klipper #1343: Pressure Advance - retract too many times?

Trontastic, 2019-03-02:
Hello guys!

I wanted to try the pressure advance, but when I enable it, it retract way too many times (~ 3 times per second on infill for example) either on the tuning test (with 0 lookahead) and on normal prints too (default lookahead time).

Should I increase the lookahead time or I can't do anything with this?

KevinOConnor, 2019-03-04:
In order for us to assist, you'll need to attach the Klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Trontastic, 2019-03-30:
Sorry for the late answer.

It's happening on the zigzag moves on the first layer.
I cancelled the print short after the layer 2 started.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3025981/klippy.log)

KevinOConnor, 2019-04-01:
That's generally how pressure advance works - it performs retractions during deceleration.  If the retractions are too much for the extruder, then the only solutions are: 1 - don't use pressure advance, 2 - use a lower printer acceleration, or 3 - use a lower pressure_advance value.  See https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md#important-notes for further details.

-Kevin

Trontastic, 2019-04-01:
Oh so it's intended. 
IMO it would be better if it only retract on the outer shells not in inner ones or infill so it would be less mechanical pressure on the printer and less chance that something goes wrong in a long print. 

I think I disable it then.

closed 2019-04-01 by Trontastic

jakep82, 2019-04-01:
Pressure advance is run-time configurable.  If you only want to run it on outer layers, you can accomplish that with scripts to enable it before outer layers, and disable it after outer layers are complete.