Klipper3d/klipper #1345: MCU 'mcu' shutdown: Rescheduled timer in the past

DBa2016, 2019-03-02:
While printing an "intentionally overextruding" model (described here: https://3dtoday.ru/blogs/osmininmaks52-gmail-com/tear-patterns-press-watson/), klipper runs into an "MCU 'mcu' shutdown: Rescheduled timer in the past" error, reproducably at 4.7%.

Klippy.log pasted here: http://paste.ubuntu.com/p/B9DBS4GvWZ/

Any suggestions?

MCU is MKS GEN L, running on 24V (if that matters) in a Tevo Tornado, with TMC2130 on all axes including extruder. All drivers run with "microsteps: 16, interpolate: True".

KevinOConnor, 2019-03-04:
In order for us to assist, you'll need to *attach* the Klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

DBa2016, 2019-03-04:
Sure, here we go. I just thought it would be easier for everyone to see the logfile on a pastebin. Here goes the zipped logfile. [klippy.zip](https://github.com/KevinOConnor/klipper/files/2928426/klippy.zip)

KevinOConnor, 2019-03-07:
Your input g-code contained a bogus move (`G1 X124.973 Y108.457 E1148.69496` after `G1 X123.931 Y109.102 E1146.40224` is too much extrusion for such a small move).  That bogus move combined with pressure advance results in a very fast extruder movement - faster than the micro-controller can generate step pulses.  The code would normally catch that and report the bogus movement, but you've disabled that check by specifying a max_extrude_cross_section.

-Kevin

KevinOConnor, 2019-03-22:
I'm going to close this as I think the original question was answered (the intentional over-extrusion mixed with high pressure advance resulted in greater extruder movement than the micro-controller could handle).

-Kevin

closed 2019-03-22 by KevinOConnor