Klipper3d/klipper #1346: Heating not stopping at set temperatur (HT-NTC100K)

vtuan10, 2019-03-03:
I recently purchased a thermistor called [HT-NTC100K](https://www.aliexpress.com/item/1M-2M-HT-NTC100K-Thermistor-Temperature-Sensor-For-High-Temperature-Filament-350-Degrees-Hotend-Kit-B3950/32832743938.html) from aliexpress.
The thermistor shows the right temperature as expected, but when I heat up my hotend, it does not stop heating at the set temperature.
I attached a video of the heating where the temperature is set to 30°C. The heater goes up to 61°C and then slowly cools down to 30°. This cycle repeats indefinitely.

What I tried:
- Changing heating element: Same results
- Also same results when changing the `sensor_type` to:
  - "EPCOS 100K B57560G104F" 
  - "ATC Semitec 104GT-2"
  - "NTC 100K beta 3950"
  - "Honeywell 100K, 135-104LAG-J01"
  - "NTC 100K MGB18-104F39050L32"
- Changing `max_power` to `0.2`: Same results
- Changing thermistor to another 100k thermistor: Heating works as expected

[klippy.log](https://github.com/KevinOConnor/klipper/files/2922599/klippy.log)
[video_heater_issue.zip](https://github.com/KevinOConnor/klipper/files/2922609/video_heater_issue.zip)

Hywelmartin, 2019-03-03:
so it works as expected... You are not melting any plastics at that temperature.. try some more realistic temperature..

KevinOConnor, 2019-03-04:
From the description It sounds like you need to run PID_CALIBRATE. Please confirm you've run through the steps described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

vtuan10, 2019-03-04:
Thanks. Using PID instead of watermark solved the issue.

closed 2019-03-04 by vtuan10

vtuan10, 2021-10-07:
> > Thanks. Using PID instead of watermark solved the issue.
> 
> Care to explain how you fixed this? I purchase this thermistor as well.

Hey there :)
This issue was quite some time ago, I'm not really sure anymore. I believe just using PID solved the issue for me.