Klipper3d/klipper #1347: Zprobe works except in delta_calibrate (query_probe works)

felixperez, 2019-03-03:
Hi!

First of all, thanks Kevin for your amazing work 👍 

I have some problem with my Smoothieboard (lpc176x), all is working, but when i use "delta_calibrate", the Z_probe no works, if I touch the probe with my hand before it reaches the base, it does not stop, it continues to descend to infinity.

If i use "query_probe" i can see if my probe is "TRIGGERED (if i touch it)" or "opened" (it's works fine)

I can't upload the .log because no log nothing about this problem.

Thanks and sorry for my bad english ^.^'

Hywelmartin, 2019-03-03:
does the PROBE command works?

Height settings wrong
Probe height wrong
offset wrong
42

It's hard to guess... the log file contains the answer..

felixperez, 2019-03-03:
> 
> 
> does the PROBE command works?
> 
> Height settings wrong
> Probe height wrong
> offset wrong
> 42
> 
> It's hard to guess... the log file contains the answer..

Hi,
I tried "probe" command and no works: " Probe triggered prior to movement "

[klippy.log](https://github.com/KevinOConnor/klipper/files/2922956/klippy.log)

felixperez, 2019-03-03:
Ok.

PROBE command works, when i  use this command and during the travel i touch the zprobe, the printer stop. 

But "delta_calibrate" no works

Hywelmartin, 2019-03-03:
Try to add some height to the position from were it starts the "probe mode" in delta calibrate 
```
[delta_calibrate]
...
...
...
horizontal_move_z: 50
#   The height (in mm) that the head should be commanded to move to
#   just prior to starting a probe operation. The default is 5.
```

KevinOConnor, 2019-03-04:
In order to use DELTA_CALIBRATE, the position_endstop settings must be accurate to within a few millimeters.  (The DELTA_CALIBRATE command makes a regular move to the horizontal_move_z position before performing a probe - if the printer does not have sane configuration values then this initial move may fail.)  Be sure to read through the docs at: https://github.com/KevinOConnor/klipper/blob/master/docs/Delta_Calibrate.md

felixperez, 2019-03-06:
Hi all, yes... 
The problem was with the position_endstop and the heigh of my printer... I thought that I could put a bigger distance from the real one and that "klipper" calibrated to the correct measure (because I did not know it).

Thanks all team :) 👍

closed 2019-03-06 by felixperez