Klipper3d/klipper #1348: printer work with hang

rizpardazan, 2019-03-03:
hello,
to day i was started my first print with klipper , but printer do his job with hangs ,
i think this is a problem with buffer,
i search the issues for this and found buffer_time_high parameter , but with height value of this (120) or low one (1) no effect.

see this video for this problem:
https://youtu.be/uxN1NpC8KRA

any idea to solve?
i use raspberry pi 3b and mega 2560 with ramps 1.5

NedalLive, 2019-03-03:
I had similar problems, but only in the curves and at high speed
can you control the Rpi CPU load please?
in SSH with the command "top" in my case it exceeded 100%

KevinOConnor, 2019-03-04:
In order for us to assist, you will need to attach the Klipper log file from the event - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Maglin, 2019-03-05:
You are using far too much micro stepping.  Turn it down to 1/16.  Your welcome.  Also, where is that log that is required?

rizpardazan, 2019-03-06:
this is the log file with SD printing,
https://ufile.io/cs0r9

rizpardazan, 2019-03-06:
> You are using far too much micro stepping. Turn it down to 1/16. Your welcome. Also, where is that log that is required?

i change it to 12800 but  still have problem

closed 2019-03-08 by rizpardazan