Klipper3d/klipper #1349: wait for a few seconds when homing

rizpardazan, 2019-03-03:
i have another problem ,
when i run g28 for homing , after homing X and Y axis then printer wait for a few seconds and then start homing Z axis 
my z axis setting :

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .00004
#endstop_pin: ^ar18
#position_endstop: 0.5
endstop_pin: probe:z_virtual_endstop
position_max: 280
position_min: -2.0
homing_speed: .66

i use bltouch for z axis.

any idea for solution ?

Hywelmartin, 2019-03-04:
A log would make it easier..

try 

`42`

or have a look at https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

BlackStump, 2019-03-04:
have a look at your step_distance 0.00004 
not sure why anyone would want 25000 steps per mm

Maglin, 2019-03-05:
@Hywelmartin has to be the best answer I've seen in a while.  It is the answer to the biggest question in the universe.  +1

closed 2019-03-06 by rizpardazan