Klipper3d/klipper #1351: Cannot reconnect after an MCU shutdown (on Duet2).

Fulg, 2019-03-03:
On a Duet2 board, after an MCU shutdown (even legitimate reasons, like `M112`), it is impossible to reconnect to Klipper via OctoPrint. Once an MCU shutdown occurs, you need to reset the Duet2 manually by pressing its reset button, then restarting the Klipper service on the Pi.

On an Arduino/RAMPS setup, connecting to it via OctoPrint will reset the MCU implicitly (via the DTR line). This does not seem to occur on Duet, I don't know if this is something that can be fixed on your side with the USB comms emulation. 

As it is now there is no way to issue a `FIRMWARE_RESTART` command on Duet after an MCU shutdown state has occurred, because you cannot connect to a shutdown MCU.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2923657/klippy.log)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/2923662/octoprint.log)

Fulg, 2019-03-03:
As a workaround, I can disable disconnects on error from within OctoPrint, but then I don't see the errors anymore unless I search for them in the terminal tab...

Fulg, 2019-03-04:
Never mind, it seems I cannot read, and I was confusing Klipper with more "traditional" firmwares. The Klipper installation docs clearly state:

_In the Settings tab, navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but stay connected to the printer" option. Click "Save"._

I am still saddened that this means I don't get to see errors as popups in OctoPrint anymore, but this is not a Klipper issue.

closed 2019-03-04 by Fulg

theopensourcerer, 2019-03-04:
FWIW, I get error popups (I think so anyway). I have the very useful Octoklipper plugin installed.