Klipper3d/klipper #1352: TMC2130 SPI config error

Gaatvul, 2019-03-03:
Hi,
I have 2No. issues with my TMC2130 stepper motor drivers.

1. I am getting an error telling me that the 'spi_bus' option is not valid for my X-axis stepper motor TMC2130 config. 

2. Whenever I physically plug my TMC2130 stepper drivers into my BIGTREETECH SKR V1.1 board, the printer gives a 'Printer is not ready' warning, however no amount of firmware or host restarts will clear this warning. I have plugged in regular A4988 stepper drivers and I get full movement from all axes.

I have attached the Klippy.log file. Any help would be greatly appreciated.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2923735/klippy.log)

_Equipment list:_
BIGTREETECH SKR V1.1 control board
4No. TMC2130 Stepper motor drivers
RPi 3A+

FHeilmann, 2019-03-04:
spi_bus and software_spi don't mix. One is for hardware spi, the other is for software spi

Gaatvul, 2019-03-04:
I see, that makes sense. Thank you very much!

KevinOConnor, 2019-03-11:
I'm going to close this as it looks like the original question was answered.

-Kevin

Gaatvul, 2019-03-11:
My apologies, I didn't make it clear that the 2No. issues are separate and don't appear to be related.

KevinOConnor, 2019-03-11:
Ah, okay - can you restate what is still open?

Gaatvul, 2019-03-11:
Issue: 
Whenever I physically plug my TMC2130 stepper drivers into my BIGTREETECH SKR V1.1 board, the printer gives a 'Printer is not ready' warning, however no amount of firmware or host restarts will clear this warning. 
I have plugged in regular A4988 stepper drivers and I get full movement from all axes.
I have also tried using software SPI and found the issue persists.

KevinOConnor, 2019-03-14:
> Whenever I physically plug my TMC2130 stepper drivers into my BIGTREETECH SKR V1.1 board

That sounds like the stepstick is being inserted incorrectly or is broken in some way.  (If the firmware doesn't start due to a hardware change, it's likely a short or other problem in the hardware.)

-Kevin

Gaatvul, 2019-03-14:
I am reasonably confident that the stepsticks are being inserted correctly and that they are functioning correctly (I have tested them on a regular RAMPS 1.4 board), however I will double check that they are functioning. Is it possible that I am have an error in the configuration? Or that there are conflicts with the chip's SPI interface, since the MOSI, MISO and CLK pins are broken out in multiple places?

KevinOConnor, 2019-03-14:
The log above and the "Printer is not ready" report are indicative of the host software not being able to communicate with the micro-controller software.  So, this indicates a problem at the micro-controller - most likely that the micro-controller is not running, but possibly that the serial/usb hardware is not functional.

-Kevin

Gaatvul, 2019-03-15:
Any idea how I could resolve this issue? A firmware reset does not seem to clear the error. Is it worth doing a fresh install?

Gaatvul, 2019-03-17:
This issue appears to have resolved itself. I am unsure what action resolved it, unfortunately.

closed 2019-03-17 by Gaatvul