Klipper3d/klipper #1353: Z offset question again

Avalonnw, 2019-03-04:
I was wondering if there is a simple way of changing Z-offset for the nozzle in config rather than through SET_GCODE_OFFSET command?
I have calibrated probe offset, and ran both delta calibration routines. The printer works perfectly now apart of the nozzle is 0.6mm too high. Is there a simple way to rectify that in config without the need to recalibrate everything again?

Hywelmartin, 2019-03-04:
Change the probe offset...in printer.cfg to the correct one, in this case 0.6 mm less than what you have
```
#[probe]
#pin: ar15
#   Probe detection pin. This parameter must be provided.
#x_offset: 0.0
#   The distance (in mm) between the probe and the nozzle along the
#   x-axis. The default is 0.
#y_offset: 0.0
#   The distance (in mm) between the probe and the nozzle along the
#   y-axis. The default is 0.
#z_offset:
#   The distance (in mm) between the bed and the nozzle when the probe
#   triggers. This parameter must be provided.
```

I usually run this macro as a part of my start code

> [gcode_macro set_delta_height]
gcode:
	M117 SET_DELTA_HEIGHT	
	G90	
	G0 F4000 X0 Y0 Z10 
	PROBE
	G0 F2000 Z3
	PROBE
	G92 Z-0.20 #your offset goes here
	G0 F4000 Z10

KevinOConnor, 2019-03-04:
Even for queries like this, we really need the Klipper log file attached (as described at https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md ) so that we have access to your config file, software version, etc.

If you have an offset of 0.6mm it sounds like the printer is not configured correctly.  See the documentation at: https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Level.md

-Kevin

closed 2019-03-22 by KevinOConnor