Klipper3d/klipper #1354: Filament Change Macro

schnello, 2019-03-04:
Hi. Yesterday I wrote the "Filament Change Macro" from Github in my printer.cfg

The original:
[gcode_macro M600]
default_parameter_X: 50
default_parameter_Y: 0
default_parameter_Z: 10
gcode:
 PAUSE
 G91
 G1 E-.8 F2700
 G1 Z {Z}
 G90
 G1 X {X} Y {Y} F3000
 G91
 G1 E-50 F1000
 G90

So basically it works. But there are two points that i do not understand:

1: M600 is called directly from the gcode file.
Octoprint goes into pause mode
It moves the nozzle on the edge of the printbed ... and move it directly back to the object (starting position)

How can i place the Nozzle on the edge ... and if i confirm with "resume" in octoprint... the printer move the the start point back?

2: The original macro makes a retract twice (5.8mm). Of course, if i completely push the filament in and then confirm with resume, the printer places a big amount of plastic on the print object.

Is it possible to restore the last known E position? To reset it to 0 doesnt work because the print will canceled.

Greetings,
Christian

Hywelmartin, 2019-03-04:
Don't combine the Octopi macros and Klipper.. it gets messy...
Configure new buttons like
[image]

The RESUME gcode will reset everything including E position 
My macros look like [this](https://pastebin.com/u2SNjdy1)
and my scripts in Octopi
[image]

schnello, 2019-03-04:
Thank you. 
I have no macros defined inside Octoprint but i can confirm. If i press the "continue" button in octoprint.. the print resume is defect (The nozzle print in the air)

If i enter "RESUME" in the command line... the printer moves down the nozzle... and continue as expected the print.

I will take now a closer look inside your macros.

Christian

[image]

closed 2019-03-04 by schnello

ArDrift, 2019-04-13:
I have the same issue with the pause_resume functionality as @schnello has described (in the first point), **the head goes back to its previous position immediately after going to the PAUSE position.** Although I've tried setting different macros (also the ones that @Hywelmartin provided) it seems it has no effect on how octoprint handles these events.
In octoprint's log it prints `Did not receive parseable position data from printer within 10.0s, continuing without it` which I think should not happen, as klipper controls the resume mechanism in itself.
I would like the PAUSE (the M600 macro) action to pause the printer, go to the desired position to be able to change the filament, and go back to its original position **only** when I give the RESUME command, as this was suggested above.
Do you have any idea on resolving this issue? Thanks for helping me out!

schnello, 2019-04-14:
Hello.

I invest some hours and i found a solution that works for me.
Important is: Dont use any functions of Octoprint!
Install the latest RC of Octoprint and add M600 as pause command and long running command.
[image]
[image]

With the macros from @Hywelmartin it works fine.

ArDrift, 2019-04-14:
Thank you, this works for me as well!

EagleeyeKai, 2020-05-01:
@Hywelmartin How do you add the custom buttons you mentioned above?

Hywelmartin, 2020-05-02:
@EagleeyeKai 
in the OctoKlipper you have the ability to add buttons
But I use custom control editor and can use sliders.. 
I have physical buttons for the most common tasks now..so I use the interface more seldom now days..  

[image]

Rhastlyn, 2021-03-16:
I like those physical buttons @Hywelmartin -- do you have any details on how you implemented them? Is that a GPIO Plugin for Octoprint or some other implementation?

Hywelmartin, 2021-03-16:
it's the funktion [button]
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#gcode_button
https://pastebin.com/jRGNMqcB
https://youtu.be/GN5M79_f_U0