Klipper3d/klipper #1355: Makefile: Wrong usage of -fwhole-program

Delsian, 2019-03-04:
In file Makefile line 38:
CFLAGS += -flto -fwhole-program -fno-use-linker-plugin
you shouldn't use -fwhole-program in this context.
As described in [http://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html](http://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html) : 

> This option should not be used in combination with -flto.

I'm trying to adopt Klipper code to another architecture and compilation fails until 
removing this key.
My suggestion to remove this key from the project.

KevinOConnor, 2019-03-06:
My interpretation of the gcc documentation is that it recommends using -flto with linker plugins instead of with -fwhole-program (but does not forbid using -fwhole-program).  Unfortunately, in this application (a small embedded program that does not use any system libraries) the linker plugins do not work well.  (The -fwhole-program option performs much better.)  So, Klipper disables the linker plugins.

-Kevin

Delsian, 2019-03-11:
Ok, I'll put this change only to the single target, where it required.
Maybe have sense to write a short comment in the makefile about linker issue in some targets.

closed 2019-03-11 by Delsian