Klipper3d/klipper #1358: Recommended host specs?

tkiapril, 2019-03-05:
Hi, I'm currently planning to build a 3d printer with integrated ARM-M0 32 bit control board and a single board computer. I am aware that you recommend using RPi3B+ per the FAQ, but I was wondering if Klipper would perform better if I used a SBC with higher specs, in particular because I have an ODROID XU3/4 with Exynos Octa-core lying around. Would you say that a RPi3B+ is the baseline? or would using a more powerful board be an overkill (with a webcam)? In terms of print quality, I am curious if using a stronger CPU would be better.

Maglin, 2019-03-05:
BLUF: NO

You clearly have never run Klipper.  It uses at most 10% of RPi3 under the most extreme loads.  Usually, it's around 2-3% of the total system load.  But I would like to have an ODROID XU4 for an SBC for my kids.

tkiapril, 2019-03-05:
You're right, I haven't tried out Klipper yet. I was just wondering because the FAQ warned about skipped steps on an RPi0, which shouldn't necessarily have tenth much power of a RPi3B+? I guess then a RPi3B+ should be plenty for Klipper+webcam setting.

mocksoul, 2019-03-05:
I disagree.

Klipper itself is not that CPU hungry, but OctoPrint is. Higher cpu just makes things quickier

konkav-js, 2019-03-21:
Hi.
I want to try the Klipper, but the space I have is quite limited. The RPi 3 A+ is smaller than B+, so I would like to ask, is A+ enough for normal operation for Klipper and OctoPrint? Or the "missing" half of the B+ RAM is going to make problems?
Thanks.

KevinOConnor, 2019-03-22:
See: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3

KevinOConnor, 2019-03-22:
Ah, to be more specific, the RPi 3 A+ has the same cpu core as the regular RPi3, so you should not have any issues with it (assuming the form factor and connectivity of the RPI A+ is sufficient).

-Kevin

KevinOConnor, 2019-03-31:
It looks like the conversation has concluded on this issue, so I'm going to close this ticket.

-Kevin

closed 2019-03-31 by KevinOConnor