Klipper3d/klipper #1360: Failure to Flash firmware onto Prusa Mk3S

tinkwithanr, 2019-03-05:
I'm working on getting the firmware flashed over onto the 2560 board, but it doesn't seem to be finished properly. Everything seems happy from the Pi/Klipper side of things, and I get the following back from the Pi:

Flashing out/klipper.elf.hex to /dev/ttyACM0 via avrdude

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "out/klipper.elf.hex"
avrdude: writing flash (21652 bytes):

Writing | ################################################## | 100% 3.83s

avrdude: 21652 bytes of flash written
avrdude: verifying flash memory against out/klipper.elf.hex:
avrdude: load data flash data from input file out/klipper.elf.hex:
avrdude: input file out/klipper.elf.hex contains 21652 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 2.79s

avrdude: verifying ...
avrdude: 21652 bytes of flash verified

avrdude: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude done.  Thank you.

So it seems to think it finished the flash properly. However the Prusa LCD gets hung up on the "Upgrading Firmware, Do Not Disconnect" screen. 

I've tried going through the process a few times now. I'm able to reflash the stock firmware fine, so I know the bootloader is functioning on the 2560. I also tried a clean install of octopi and klipper with the same results.

KevinOConnor, 2019-03-06:
Nothing in the above is indicative of a failure.  You should continue with the installation steps as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

If you do encounter an error then you'll need to attach the klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

tinkwithanr, 2019-03-08:
My mistake, you were correct! I was expecting the prusa to confirm a successful flash, 

It's up and running now, thanks for your help!

closed 2019-03-08 by tinkwithanr