Klipper3d/klipper #1361: 3dtouch tronxy x5s

jstankowski88, 2019-03-05:
I have a stock tronxy x5s and would like to add a geeetech 3dtouch probe to it. I'm struggling because I can't seem to find what the pin numbers are because the tronxy x5s board seems largely undocumented. I have a pin 27 breakout board I was trying to use, I also thought about trying to use the miso pin.

blackyarr, 2019-03-05:
https://www.thingiverse.com/thing:3439261

maybe this will help you a little

jstankowski88, 2019-03-06:
Thank you for the link. I can't believe I missed that.

jstankowski88, 2019-03-06:
Unfortunately still no luck, I think I gave it wired correctly but I still can't deploy the pin with the debug commands. Any suggestions would be greatly appreciated.

KevinOConnor, 2019-03-07:
In order for us to assist the Klipper log file must be attached to this issue - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Beyond that, the only other suggestions would be to look at the example config files in the config/ directory.

-Kevin

closed 2019-03-22 by KevinOConnor