Klipper3d/klipper #1363: TMC2130: Missing steps when reaching "Stealthchop threshold"?

pienjo, 2019-03-05:
This has been haunting me for a while, I worked around the issue by printing larger objects at a very low speed, but decided to investigate a bit further.

My printer (FLSun "large delta" printer, MKS Gen-L board, TMC2130 drivers on the 3 main axis) seems to be badly skipping steps when the "Stealthchop threshold" is reached. It doesn't skip steps when StealthChop is entirely disabled, or when the stealthchop threshold is set to a very large number. It seems to mainly have this issue near the edge of the bed (possibly because the steprate is higher there, due to the delta geometry)

 I uploaded three small videos to youtube to demonstrate. In each, I executed the same short G-code file, which goes back and forth between X0 Y100 Z10 and X0 Y-100 Z10, using an ever increasing feedrate (initially 2000 mm/min, increases in 1000 mm/min steps)

[https://www.youtube.com/watch?v=-6tkMb2-Ios](url)  -  "Stealthchop threshold 200", I aborted the print to prevent a head crash. I verified that it is the stepper skipping, not the belts or pulleys.
[https://www.youtube.com/watch?v=ypL25iaRQXk](url) - "Stealthchop threshold 3000. No missed steps.
[https://www.youtube.com/watch?v=JYEikuGlsIk](url) - "stealthchop threshold 0", no missed steps. 

What can I do to address this? 

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2932446/klippy.zip)

KevinOConnor, 2019-03-06:
This is a limitation of the tmc drivers.  There's lots of information in the tmc2130 spec about this.  In short, the drivers are known to skip steps when transitioning from "spreadcycle" mode back to "stealthchop" mode at high speed.  The spec describes lots of tuning mechanisms, but to the best of my knowledge, the only real solution is to set the stealthchop threashold above the maximum velocity of the stepper or to a small value (eg, 5 or less).

-Kevin

pienjo, 2019-03-06:
Well, nothing to be done then :-/

closed 2019-03-06 by pienjo

nettunoroberto, 2019-04-26:
I also have the same problem on my hypercube. A question, but if I leave the conf fixed on stealthchop mode what problems could I face? can't I print at high speeds?

elcojacobs, 2019-05-14:
Okay, this took me a long time to dial in but I found settings that work for me by experimenting a lot and by digging through the datasheet of the TMC2130.

The drivers can be configured to be a lot more aggressive than the default settings, even in stealthchop mode.
The PWM_GRAD can change how quickly the PWM value can ramp up and down. Pressure advance requires a very quick switch in direction and the default setting of 4 is not quick enough for my extruder.
I have a BMG extruder with 1:3 gear ratio, which means my stepper has to move 3x quicker even.

The strength and responsiveness of the stepper depend on the current and the aggressiveness of the algorithm. Too aggressive and too high current and the drivers quickly overheat.

- To combat overheat I reduced the current of the extruder stepper to 0.5A. Hold current at the same value to always be ready to switch direction instantly.
- I am using an MKS gen L board and printed the Teaching Tech case for it. I moved all my steppers up one slot. None in X, and extruder in extruder 1. This moves the extruder stepper very close to the fan so I can push it hard.
- I set PWM_GRAD to 10 for all my steppers. They are still completely silent when extruding lines, but I can do a 180 degree turn so fast hear the extruder knocking, without skipping steps.
- I tried higher currents and higher PWM_GRAD values for the extruder, but the driver quickly overheats.
- I have set the stealtchop threshold to 5000. No need to switch to spread cycle with these settings. Switching between the two should be avoided because the motor is not responding well during the switch. From the datasheet I get the idea that it is designed for increasing speed gradually. They state a 100ms wait time. So either run with pressure_advance at 0 or at a super high value.

My extruder now has no problem with a 60mm/s retract at 1.5mm. Note that the stepper is moving at 180mm/s for that due to the gear ratio of the BMG!
I dialed it down a bit to 50mm/s retract and 40mm/s prime afterward.

Even when printing a 4 pilar retraction test, the driver doesn't overheat. This is confirmed with DUMP_TMC stepper=extruder. If that shows an overheat, the driver will have temporarily shut down and you will have missed some extrusion. This can be seen in the prints, as soon as it overheats, you'll miss a line.

I spent a lot of time tuning my other settings for printing PETG. I have pressure advance at 0.45 now and I can print without any stringing, with a bowden tube!

Thanks a lot for klipper. I hope my input helps others! I was told the TMC2130's were not really suitable for the extruder, but they are so configurable. 
Just takes time to find the sweet spot. I love them now!

```
[tmc2130 stepper_x]
cs_pin: ar59
microsteps: 16
run_current: 0.700
hold_current: 0.5
stealthchop_threshold: 5000
driver_SGT: 3
driver_PWM_GRAD: 10
diag1_pin: ^!ar3

[tmc2130 stepper_y]
cs_pin: ar63
microsteps: 16
run_current: 0.700
hold_current: 0.5
stealthchop_threshold: 5000
driver_PWM_GRAD: 10
driver_SGT: 3
diag1_pin: ^!ar14

[tmc2130 stepper_z]
cs_pin: ar40
microsteps: 16
run_current: 0.700
hold_current: 0.5
driver_PWM_GRAD: 10
stealthchop_threshold: 5000

[tmc2130 extruder]
cs_pin: ar42
microsteps: 16
run_current: 0.5
hold_current: 0.5
driver_PWM_GRAD: 10
stealthchop_threshold: 5000

[stepper_x]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
# endstop_pin: tmc2130_stepper_x:virtual_endstop
endstop_pin: ^ar3
position_endstop: 0
position_max: 235
[... 165 lines cut ...]
#   trigger a split. Default is .025.
#move_check_distance: 5.0
#   The distance (in mm) along a move to check for split_delta_z.
#   This is also the minimum length that a move can be split. Default
#   is 5.0.
#mesh_pps: 2,2
#   A comma separated pair of integers (X,Y) defining the number of
#   points per segment to interpolate in the mesh along each axis. A
#   "segment" can be defined as the space between each probed
#   point. The user may enter a single value which will be applied
#   to both axes.  Default is 2,2.
#algorithm: lagrange
#   The interpolation algorithm to use. May be either "lagrange"
#   or "bicubic". This option will not affect 3x3 grids, which
#   are forced to use lagrange sampling.  Default is lagrange.
#bicubic_tension: .2
#   When using the bicubic algorithm the tension parameter above
#   may be applied to change the amount of slope interpolated.
#   Larger numbers will increase the amount of slope, which
#   results in more curvature in the mesh. Default is .2.
#relative_reference_index:
#   A point index in the mesh to reference all z values to. Enabling
#   this parameter produces a mesh relative to the probed z position
#   at the provided index.
```

elcojacobs, 2019-05-14:
I was still able to overheat the driver with the settings above by printing gyroid infill with 1.5mm line distance. I changed the pressure lookahead time to 0.02 instead of 0.01 and did not have errors. Even set square corner velocity back to 5.

elcojacobs, 2019-09-29:
Because this was referenced, here are the settings I have been using for a long time now, with very silent and good looking prints.
Please note that I am not using the standard pins, see above.
This is with a bondtech like dual drive geared extruder and an MKS gen L main board.
I'm not using the standard pins so I could move the drivers closer to the front fan.

<details>

``` 
# This file contains common pin mappings for the 2018 Creality
# Ender 3. To use this config, the firmware should be compiled for the
# AVR atmega1284p.

# Note, a number of Melzi boards are shipped with a bootloader that
# requires the following command to flash the board:
#  avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex
# If the above command does not work and "make flash" does not work
# then one may need to flash a bootloader to the board - see the
# Klipper docs/Bootloaders.md file for more information.

# See the example.cfg file for a description of available parameters.

[tmc2130 stepper_x]
cs_pin: ar59
microsteps: 16
run_current: 0.700
hold_current: 0.5
stealthchop_threshold: 5000
driver_SGT: 3
driver_PWM_GRAD: 10
diag1_pin: ^!ar3

[tmc2130 stepper_y]
cs_pin: ar63
microsteps: 16
run_current: 0.700
hold_current: 0.5
stealthchop_threshold: 5000
driver_PWM_GRAD: 10
driver_SGT: 3
diag1_pin: ^!ar14

[tmc2130 stepper_z]
cs_pin: ar40
microsteps: 16
run_current: 0.700
hold_current: 0.5
driver_PWM_GRAD: 10
stealthchop_threshold: 5000

[tmc2130 extruder]
cs_pin: ar42
microsteps: 4
run_current: 0.5
hold_current: 0.5
driver_PWM_GRAD: 8
stealthchop_threshold: 5000

[stepper_x]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
# endstop_pin: tmc2130_stepper_x:virtual_endstop
endstop_pin: ^ar3
position_endstop: 0
position_max: 235
homing_speed: 50
second_homing_speed: 5

[stepper_y]
[... 214 lines cut ...]
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*#
#*# [bed_mesh ender3]
#*# points =
#*# 	  0.057500, 0.052500, 0.035000
#*# 	  0.012500, -0.007500, 0.000000
#*# 	  0.055000, 0.047500, 0.042500
#*# x_count = 3
#*# y_count = 3
#*# min_x = 115.0
#*# max_x = 215.0
#*# min_y = 61.0
#*# max_y = 201.0
#*# x_offset = -50.0
#*# y_offset = -16.0
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
```

</details>

robustini, 2019-12-17:
>     * I tried higher currents and higher PWM_GRAD values for the extruder, but the driver quickly overheats.
> 

Here a value near 10 produce a little vibration on XYZ axis (delta printer with TMC2130), visible on the printed object.