Klipper3d/klipper #1364: TMC2130 & RAMPS 1.4 & LCD Help needed please

quintinjvr, 2019-03-06:
Hello all

Firstly, @KevinOConnor and community, thanks for all the work that goes into Klipper. I recently moved over from Repetier to Klipper and it was the best decision ever, print quality is amazing!

I've decided to upgrade my printer from the DRV8825 to the **TMC2130** drivers on all my axes (X,Y,Z & E0), but I find it a bit challenging to find clear information or example config files with accurate values for my setup with SPI etc.

Can anyone please assist with a complete config file indicating the correct pins and values for the TMC2130's on a RAMPS 1.4? I would also like to continue using my RepRapDiscount 128x64 Full Graphic Smart (st7920), but I see that you need to reassign pins. I am unsure how to approach this with Klipper as well

I used the wiring below from Toms3D [https://toms3d.org/2017/12/09/tmc2130-guide/](url)

[image]
[image]
[image]

But the software side if things are all done with Marlin.

Assistance will be much appreciated as I am somewhat new to Klipper.

Thank you in advance and keep well.

quintinjvr, 2019-03-06:
Sorry - Log file attached
[klippy.log](https://github.com/KevinOConnor/klipper/files/2935863/klippy.log)

noud-github, 2019-03-10:
what is the problem you experiance? i run display and 2 z steppers with tmc2130 from ramps, i only use aux2 pins for chip select and the 3 spi from aux 3,  (i use 3 mcu's)  aux2 has analog and digital pins you can use all with the digital pin names with klipper. finding the digital names is not that hard with the ramps pin file from marlin, and a pinout image from the ramps

quintinjvr, 2019-03-11:
Thanks for your reply @noud-github .

I managed to figure out the pinouts and got the drivers to work.  But I cannot get the SPI settings to work in order to fine tune the drivers. I've looked through a few other klippy.log files from other users and noticed that very few people use it (as it appears to be optional). Could you please advise on what SPI Bus and Speed to use? Do I use the same pin numbers that the drivers are connected to for spi_software pins (slck, mosi, miso)?

```
#cs_pin:
#   The pin corresponding to the TMC2130 chip select line. This pin
#   will be set to low at the start of SPI messages and raised to high
#   after the message completes. This parameter must be provided.
#spi_bus:
#spi_speed:
#spi_software_sclk_pin:
#spi_software_mosi_pin:
#spi_software_miso_pin:
```
Any assistance will be much appreciated.

KevinOConnor, 2019-03-11:
Under normal circumstances, none of the optional parameters would be specified.  Only the cs pin would be specified.

If you've gotten further along, you should attach the log from the latest attempt so others can see what you attempted.

-Kevin

noud-github, 2019-03-11:
as far as i know atmega only exposes 1 spi head so if it works you use te right one ;)
speed i use default. and since you have 6 slaves. 5x tmc 1x display. and de dislpy is know for not playing nice with the spi protocal i recomend default
slk 52
miso 50
mosi 51 
cs_pin is one from the aux2 d63 d64 d65 d66  d40 d42 d44 can be used

noud-github, 2019-03-11:
the options are there for mcu's with more spi interfaces.

noud-github, 2019-03-11:
one last tip use gcode  DUMP_TMC STEPPER=stepper_x if the replay is not all 0 or F the spi comunication is up and runnig

quintinjvr, 2019-03-11:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2953938/klippy.log)

Thank you so much for clarifying @KevinOConnor  & @noud-github, all makes sense now.
I also gave it a go with DUMP_TMC STEPPER=stepper_x and all is ok.

Now that everything is sorted with the drivers, SPI and LCD, the next thing on the list will be limits and BLTouch :-)

I also have something else that I would like to explore and perhaps open a new thread for it, but I have a [Nextion 3.2 LCD Touch display](https://www.itead.cc/wiki/NX4827T043#Download_Datasheet) that works with UART (serial) and think it would be quite cool to get it to work. Do you perhaps know if there is anything available to get me started on that or any support for it on Klipper?

Here is a [link](https://github.com/itead/ITEADLIB_Arduino_Nextion) to the Nextion-Arduino library for it.
And [link ](https://nextion.itead.cc/resources/documents/instruction-set/)to Nextion instruction set.

noud-github, 2019-03-12:
great to hear, it is working  
the display thing is off topic,  the "uart" klipper exposes by default is /tmp/printer software but you need hardware for a display so yes make a new topic for that

KevinOConnor, 2019-03-22:
I'm going to close this as I think the original issue was resolved.  The display question is separate, and to the best of my knowledge is not supported currently.

-Kevin

closed 2019-03-22 by KevinOConnor