Klipper3d/klipper #1365: TMC 2208 on Fystec F6, Unable to read register IFCNT

szaghi, 2019-03-06:
Hi all,

thank you very much for your great work, it is really appreciated.

I am not able to make working my new Fysetc F6 board with TMC2208 in UART mode, please see below my klippy log.

I have tried to dig deeper my self reading older issues, but I have not understand what I am doing wrong. The only doubt I have (and not yet able to test is) is related to the board powering, I have connected it only to the raspberry via USB without 12V PSU supply.   Is the PSU powering relevant for driver register reading?

I am a newbie (although @SimoneBnc is kindly trying to help me) thus surely I am doing some very silly mistake.

Thank you for any hints and apologize my self if this was a well known issue.

```ini
===== Config file =====
[stepper_x]
step_pin = PF0
dir_pin = PF1
enable_pin = !PD7
step_distance = .0125
endstop_pin = ^PK1
position_endstop = 0
position_max = 300
homing_speed = 50

[tmc2208 stepper_x]
uart_pin = PG3
tx_pin = PJ2
microsteps = 16
run_current = 0.7
hold_current = 0.7
stealthchop_threshold = 250

[stepper_y]
step_pin = PF6
dir_pin = PF7
enable_pin = !PF2
step_distance = .0125
endstop_pin = ^PJ1
position_endstop = 0
position_max = 300
homing_speed = 50

[tmc2208 stepper_y]
uart_pin = PJ3
tx_pin = PJ4
microsteps = 16
run_current = 0.7
hold_current = 0.7
stealthchop_threshold = 250

[stepper_z]
step_pin = PL6
dir_pin = !PL1
enable_pin = !PF4
step_distance = .0025
endstop_pin = ^PB6
position_endstop = 0
position_max = 400
position_min = -1

[tmc2208 stepper_z]
uart_pin = PE2
tx_pin = PE6
microsteps = 16
run_current = 0.7
hold_current = 0.7
stealthchop_threshold = 100

[extruder]
step_pin = PA4
dir_pin = !PA6
enable_pin = !PA2
step_distance = .0025
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PE3
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PK4
control = pid
pid_kp = 29.662
pid_ki = 2.326
pid_kd = 94.549
min_temp = 0
max_temp = 320

[tmc2208 extruder]
uart_pin = PJ5
tx_pin = PJ6
microsteps = 16
run_current = 0.9
hold_current = 0.9
stealthchop_threshold = 250

[heater_bed]
heater_pin = PH5
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PK7
control = pid
pid_kp = 74.536
pid_ki = 1.206
pid_kd = 1151.587
min_temp = 0
max_temp = 130

[fan]
pin = PL5

[heater_fan my_nozzle_fan]
pin = PL4
shutdown_speed = 0
fan_speed = 0.9

[heater_fan my_control_fan]
pin = PL3
shutdown_speed = 0

[mcu]
serial = /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100
=======================
Loaded MCU 'mcu' 77 commands (v0.7.0-226-gb658940-20190227_130817-octopi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 SOFT_PWM_MAX=256 STEP_DELAY=0 SERIAL_BAUD=250000 ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 MCU=atmega2560 STATS_SUMSQ_BASE=256
Configured MCU 'mcu' (612 moves)
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-226-gb658940'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
=============== Log rollover at Wed Mar  6 10:53:16 2019 ===============
Unable to read tmc2208 'stepper_x' register IFCNT
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Unable to read tmc2208 'stepper_x' register IFCNT
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Unable to read tmc2208 'stepper_x' register IFCNT
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Unable to read tmc2208 'stepper_x' register IFCNT
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Unable to read tmc2208 'stepper_x' register IFCNT
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Unable to read tmc2208 'stepper_x' register IFCNT
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

SimoneBnc, 2019-03-06:
Have you soldered the pads under the driver pcb and set the correct jumper configuration as specified in the fysetc F6 wiki page?

szaghi, 2019-03-06:
@SimoneBnc 

Hi Simo, yes, I hope so, I followed your last video where the guy suggest to solder the 3 pads near the chip, I having the Fystec TMC2208 v1.2 with 3 pads instead of 2. The jumper configuration should be right, I have shorted the 2 left pin below the drivers.

szaghi, 2019-03-06:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2936140/klippy.log)

Stephan3, 2019-03-06:
@szaghi 
i am not sure but the logfile looks incomplete to me.

szaghi, 2019-03-06:
Hi @Stephan3 let me know what I have to do to create a complete log, I have just followed the indication to do a M112 command after I saw the *unable to read...* issure

FHeilmann, 2019-03-06:
I'm not familiar enough with the tmc2208 in particular. But in case you haven't considered it, think about that the communication with your drivers might not be working at all and the "IFCNT" register is simply the first one that fails to be read.

-Florian

szaghi, 2019-03-06:
Hi @FHeilmann thank you very much for your insight. In case the communication is not working at all what do you suggest? What can I do for checking? I have some LV8729 drivers

Stephan3, 2019-03-06:
@szaghi did you solder the pad on the buttom? show us a picture.

szaghi, 2019-03-06:
@Stephan3 yes, maybe my soldering skill is the problem... here is my very bad *pond bridge*

[image]

Stephan3, 2019-03-06:
@szaghi Ia ma not Sure about those clones as i use usaly watterotts. Regarding to this you did the soldering wrong?
https://www.instructables.com/id/UART-This-Serial-Control-of-Stepper-Motors-With-th/
https://cdn.instructables.com/FF5/IH73/JHATJPEZ/FF5IH73JHATJPEZ.LARGE.jpg?auto=webp&width=350

KevinOConnor, 2019-03-06:
> Is the PSU powering relevant for driver register reading?

Yes.  In all common configurations, the TMC drivers are powered from "motor power".  So, if the external power supply is not enabled then one will not be able to communicate with the drivers.  Be sure to bring up the external power supply first, then issue a RESTART command to have Klipper initialize the drivers.

-Kevin

JamieAKing, 2019-03-06:
Good new ...The soldering looks good (by good I mean the same as mine,) so the chip should communicate with Klipper. 
Bad news .... I have the same board and drivers and have had very limited success.
I hope you have better luck than I’m having, if you do could I ask that you post your stepper configs so that I may compare. 

Steps taken so far....
Z is working. I had to set interpolation to false and micro to 16. I also followed the steps in issue #196 to setup auto_grad and stealthchop threshold
Extruder is only reliable if above is done AND stealthchop is disabled (set to 0), noisier but not drv8825 noisy. I did try silly high stealthchop_threshold 100000 but still not reliable.
I have yet to find a xy configurable that is reliable. 
Patience is at an all time low and I’m currently awaiting some 2130s to replace the drv8825 that are currently running the XY
Hope this helps a little 
Best of luck 
Regards 
Jamie

szaghi, 2019-03-06:
Hi Jamie, thank you very much for your insight. Maybe tomorrow I'll be able to power on the board (the printer is busy, I have to stop printing for a while to power the F6). Anyhow I am really a newbie regarding stepper configuration, I am simply following Simone that had success in using tmc2208+f6+klipper.

I'll post my progress here, I hope tomorrow.

Cheers

Mamath2000, 2019-03-16:
Hi friend, 
yes  : TMC2208+f6+klipper, it works like a charme since at least 6 mounth.
how have you configure jumper on the board. perhaps your problem is just a F6 setup issue. 

push a picture

KevinOConnor, 2019-03-22:
I'm going to close this as I'm guessing the original issue was resolved (motor power needs to be supplied).

-Kevin

closed 2019-03-22 by KevinOConnor