Klipper3d/klipper #1369: Differennt homing_override for z and xy

kakou-fr, 2019-03-07:
I need two different homing_override for Z and XY.

I used the first one until today (capacitive probe), but now i want that the nozzle move Z+15 when G28 X0 Y0.

I tried this, but only the second is applied : 

`[homing_override]
set_position_z: 10
axes: z
gcode:
    G90   
    G1 Z15 F600
    G28 X0 Y0
    G1 X110 Y10 F3600
    G28 Z0

[homing_override]
set_position_z: 10
axes: xy
gcode:
    G90  
    G1 Z15 F600
    G28 X0 Y0`

jakep82, 2019-03-07:
If you want the homing override to apply to all G28 commands, then delete the second override, and change the first one to

`[homing_override]
set_position_z: 10
axes: xyz
gcode:
    G90   
    G1 Z15 F600
    G28 X0 Y0
    G1 X110 Y10 F3600
    G28 Z0`

kakou-fr, 2019-03-07:
I need two differents homing, because i don't want that the sensor try to probe the bed when i do a G28 X0 Y0

jakep82, 2019-03-07:
So how are you homing Z?  I'm not sure what you're trying to accomplish.

kakou-fr, 2019-03-07:
I use G28 for homing, or just G28 X0 Y0 only for X/Y or just G28 Z0 for only Z.
With only the first homing_override, G28 or G28 Z0 lift the nozzle +15mm. but not  G28 X0 Y0.
So i try to find a way to also lift nozzle when i do a G28 X0 Y0.

jakep82, 2019-03-07:
I'm not sure if this is possible.  The homing_override should home all axes per [example-extras.cfg.](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L1099)

kakou-fr, 2019-03-07:
In fact, i search a way to always lift my nozzle before homing any axis in order to prevent a collision with my bed clips.

jakep82, 2019-03-07:
You can do that by using the homing_override I provided, but it will also home Z when you home X or Y.

kakou-fr, 2019-03-07:
yes, i why i need something else :)

jakep82, 2019-03-07:
The only other option would be something like a macro using a force_move before homing X and Y.  But that wouldn't work if you invoke G28 X0 Y0.  It would be something home_XY, and it would force move the bed down before homing X and Y.  Otherwise I think you're stuck using the homing_override which needs to home every axis.

kakou-fr, 2019-03-07:
I think i cannot write a macro for "G28 X0 Y0" only for G28. And, fo example, in octoprint, when you click on the homing XY icon it use :

```
Send: G91
Send: G28 X0 Y0
Recv: ok
Send: G90
```

And i don't find any parameter to lift Z when homing.

jakep82, 2019-03-07:
Correct, you can't write a macro for G28.  You would need to name it something like home_XY, and you would need to use a force_move to lift the nozzle before homing X and Y.  Is there a reason you want to home X and Y without homing Z?

kakou-fr, 2019-03-07:
just because i need it :)
or maybe because i use the XY homing in octoprint during testing and want to prevent crash into bed clips.

jakep82, 2019-03-07:
The safe way to accomplish that is to use the homing_override for every axis.  As noted above though, this does mean that you will need to home Z, or at least that's my understanding of how the override works.

kakou-fr, 2019-03-08:
dear bot, please find the klippy.log in attachment, even if this will not solve my problem
[klippy.log](https://github.com/KevinOConnor/klipper/files/2946815/klippy.log)

KevinOConnor, 2019-03-11:
I'm not sure there is anything further to add to this - only one homing_override is supported.  As mentioned earlier, if one is willing to modify their existing g-code scripts then it is also possible to make a HOMEXY gcode_macro.

-Kevin

KevinOConnor, 2019-03-22:
I'm going to close this as I think the original question was answered (only one homing_override may be specified and it must home all axes).

-Kevin

closed 2019-03-22 by KevinOConnor