Klipper3d/klipper #1370: Multi Stepper Z Axis motors rotate same direction regardless move value (positive or negative)

Lazurx, 2019-03-07:
Hi. Im having weird issue with multi stepper axis. Im at the point of testing Z axis assembly. Ive set multi stepper on Z axis on my COREXY assigning second stepper as unused EXT2 pin layout. When trying to control Z axis movement in Octoprint steppers work and synced but no matter positive or negative movement steppers are rotating the same way (CW). It happens when i click arrows on CONTROL tab or when i put gcode command myself. In this particular case bed is moving only UP. I added homing override to the config file setting z position to 0. Only one endstop is assigned for original Z stepper no endstop for additional Z1 stepper. Im sorry if its some newbie mistake but I have no idea why is this happening. Also couldnt find information about this.

megablue, 2019-03-08:
sounds like you didn't wire the DIR pins correctly, without the proper signals, stepper drivers can only assume the default/or whatever signals it is receiving from the wrong wiring.

Lazurx, 2019-03-09:
Drivers are pollolu step sticks drv8825 so Theres no room for such mistake. I could have had motor drivers wired wrong but it rotates correctly Also this wiring works fine with other "stepper channels" where it doesnt happen. This issue only happens on ext1 nad ext2 channels. I checked Z channel and ext0 also and they are fine. It might have something to do with pin mapping provided withiwithin config files. I used generic-rumba config in this case(this is my MCU board, chinease rumba+ clone in Fact Magdę they messed up something)

Lazurx, 2019-03-12:
I discovered that dir_pin of the EXT1 and EXT2 drivers are incorrectly set in generic_rumba.cfg file. I compared them with reprap wiki https://reprap.org/wiki/RUMBA and it turns out that dir_pin for EXT1 is ar25 (cofig file says ar 3) and EXT 2 is 28 (config says ar6) I checked it on EXT1 only (lack of time) and steppers are rotating the same direction so i think it is solved. Maybe you should recheck this config file regarding PIN mapping?

jakep82, 2019-03-12:
This does look like a mistake in the config file.  ar3 and ar6 are actually the heater pins for E1 and E2.  The config currently calls for ar9 as the heater pin on both of those, but that's actually the heater pin for the bed.

jakep82, 2019-03-12:
Corrected config below.

```#[extruder1]
#step_pin: ar26
#dir_pin: ar28
#enable_pin: !ar27
#heater_pin: ar3
#sensor_pin: analog14
#...

#[extruder2]
#step_pin: ar29
#dir_pin: ar28
#enable_pin: !ar39
#heater_pin: ar6
#sensor_pin: analog13```

Lazurx, 2019-03-12:
Ok im glad i made something useful here :D

KevinOConnor, 2019-03-14:
Okay, thanks.  Commit 378402b4.

-Kevin

closed 2019-03-22 by KevinOConnor