Klipper3d/klipper #1371: tmc2130 way to change any field registers via config fields

blairq, 2019-03-08:
im sorry im new but i cant find a  a way to change tmc2130 field registers via config settings 

im looking to change some of the CHOPCONF field registers .  or any register .

i have a large stepper on z  and its making some nasty noise.

jakep82, 2019-03-08:
Every available setting is listed in example-extras.cfg.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L891

blairq, 2019-03-08:
can you label this as enhancement ? 
posting the log is quite useless as the  entire system is working almost fine . and this issue is not about something i wanna fix using current settings or configurations. 

off course i have already read the docs.  even the code. but my python-fu is quite rusty and i was looking for some educated answer about changing the tmc2130 registers via config settings . not just hacking my way thru the code like a berserker.

FHeilmann, 2019-03-08:
did you check the link that @jakep82 posted? You just have to put some additional lines into your printer.cfg to set the fields. Each config parameter is named after the field it controls

jakep82, 2019-03-08:
It would appear you didn't read the link I posted which contains exactly what you requested; configuration options for the 2130.

closed 2019-03-16 by klipper-gitissuebot