Klipper3d/klipper #1374: hang problem also with virtual sd printing

rizpardazan, 2019-03-08:
i have enabled virtual sd and start printing with M24 /x.gcode
but i still have hang problem while printing ,
my log file :
[klippy.log](https://github.com/KevinOConnor/klipper/files/2946582/klippy.log)

https://ufile.io/h6qvp

Micro step setting are 3200
Raspberry Pi 3b with Arduino mega and ramps
Motor drivers are external leadshine and moons driver

axis setting :

[stepper_x]
step_pin: ar54
dir_pin: !ar55
enable_pin: !ar38
step_distance: .01425
endstop_pin: ^ar3
position_max: 310
homing_speed: 20.0

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .01272
endstop_pin: ar14
position_endstop: 330
position_max: 330
homing_speed: 20.0

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .0001
#endstop_pin: ^ar18
#position_endstop: 0.5
endstop_pin: probe:z_virtual_endstop
position_max: 280
position_min: -2.0
homing_speed: .66

and gcode file is :
https://ufile.io/mnlvd

trevjonez, 2019-03-08:
what is the name of the original file on disk? I have issues when the file has any spaces in the name. 

```
Starting SD card print (position 0)
virtual_sdcard seek
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 136, in work_handler
    self.current_file.seek(self.file_position)
AttributeError: 'NoneType' object has no attribute 'seek'
Unable to seek file
```

rizpardazan, 2019-03-08:
> Starting SD card print (position 0)

file name is c.gcode no spaces

rizpardazan, 2019-03-09:
problem solved

closed 2019-03-09 by rizpardazan

trevjonez, 2019-03-09:
What was the issue and solution?