Klipper3d/klipper #1375: Previous MCU 'mcu' shutdown: Missed scheduling of next pwm event

obviator, 2019-03-08:
I previously had Klipper working OK using a Raspberry Pi Zero by printing from the virtual SD Card but wanted a beefier setup so I installed Octoprint and Klipper to a virtual machine.

Everything seems to work fine and the print starts but then I get MCU errors quite soon after starting. It is not always at the same point but if I was pattern spotting it seems to be around the time the printing speed picks up after the initial layers.

I was initially and consistently getting "MCU timer to close" errors. I have since changed the baud rate down to 115200 (from 250000) on both host and processor in case that helps.

The first print I tried after this immediately errored with the title message and when I restarted it I got this pwm scheduling error after the first few layers.

Log attached after the pwm error. I can try to recreate the "timer too close" error if that log would be useful.

I realise my setup is probably unusual but any suggestions welcome!

Host: VMWare ESXi
VM: Raspbian Stretch x86
Printer: Ender 3 (atmega1284p)

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2947181/klippy.zip)

KevinOConnor, 2019-03-09:
The log indicates that there is a continuing series of retransmissions between host and mcu.  (The bytes_retransmit count keeps increasing.)  It's unclear if that is due to the VM, due to bad wiring, or due to something else.  It appears that these transmission errors are the root cause of the problem.

-Kevin

obviator, 2019-03-10:
Thanks for checking it out.

If there's no way to home in on what might be causing the re-transmits then I'll continue to experiment.
Happy for the issue to be closed although not sure if that will prevent me updating it with any findings.

KevinOConnor, 2019-03-22:
Okay - I'm going to close this.  Feel free to update this ticket if you have additional findings.

-Kevin

closed 2019-03-22 by KevinOConnor