Klipper3d/klipper #1376: [Feature Request] Dual Z Motor Alignment with two endstops, possible also when using Z-Probing

HWiese1980, 2019-03-09:
I would like to request a feature. There are printers including mine that use a print **head** moving solely along the Z-axis and a moving print **bed** for X and Y (so called bed pushers). Among such there are printers including mine that use two stepper motors for each side of the Z-axis. This has the downside that the two sides can get misaligned when only one of the lead screws rotates, for whatever reason. 

The proposed feature is obviously as follows: make it possible to realign the two sides by moving the Z-beam (upwards in my case) until both sides of the axis trigger their corresponding endstop switch. 

In details, as I imagine it for my printer: there are two endstop switches at high-Z, one on each side of the Z-frame, close to the corresponding lead screw. If alignment move is started, both Z-motors move the Z-beam along +Z. Once the first endstop is triggered, the motor that belongs to that side of the printer stops. The other keeps moving until the second endstop is triggered as well. When both endstops are triggered, the Z-beam can be considered aligned. 

Using high-Z in this case is for those like me who use a Z-probe as low-Z endstop. In this case an additional low-Z alignment endstop setup with switches would not be possible. Of course my feature request should also be able to trigger at low-Z. 

If I remember correctly, **part of this** is already possible today if you don't use a Z-probe. Two endstops for one axis at either low or high value of the corresponding axis is perfectly possible if I'm not mistaken. But it's essentially three+ in my case/proposal, i.e. one low-Z endstop and **additionally two high-Z endstops for a slightly different purpose. 

So... what do you think? A good feature? Or in fact already possible and I overlooked something?

HWiese1980, 2019-03-09:
This is a feature request. How would a log help?

Hywelmartin, 2019-03-09:
https://github.com/KevinOConnor/klipper/blob/9466c2d66cce7290bb73edcd7123d0b3a04a9021/config/example-extras.cfg#L309

HWiese1980, 2019-03-09:
This does not work for Z-beam bed pushers as, with a perfectly levelled bed, the Z-position does not change during X/Y moves.

HWiese1980, 2019-03-09:
I've added a picture of what I mean

HWiese1980, 2019-03-09:
To clarify, @Hywelmartin, this is not a Prusa like printer where the X- and Z-axes are on the beam and the Y-axis is on the bed. This is a printer (Velleman K8200) where X and Y are on the bed and Z **only** is on the beam. 

So I'm basically asking for an extension of the available Z_TILT feature but for printers of my configuration.

Hywelmartin, 2019-03-09:
I know the type, of printer..

If you configure 2 Z steppers with corresponding  endstops at each axis it will move each stepper until the endstop triggers [Link](https://github.com/KevinOConnor/klipper/blob/9466c2d66cce7290bb73edcd7123d0b3a04a9021/config/example-extras.cfg#L487)
So the gantry would be "level" after a home either at the top or bottom( where you place the endstops). then you can use your probe to set the bed surface "0"

HWiese1980, 2019-03-09:
That sounds promising. Is it possible to have the two endstops at high-Z (like about Z150 in my case) and make the gantry move **up** while adjusting gantry tilt and **down** for probing?

Hywelmartin, 2019-03-09:
the "adjustment" will happen when it hits the endstops and then it's just to probe the bed(no virtual endstop probe thingy)

So yes

HWiese1980, 2019-03-09:
Alright, and how do I move the gentry upwards? Manually by a G0/G1 command (that requires Z-min homing beforehand)? Or can I with the G28 command choose on-the-fly which trigger to use Z-min or Z-max?

Because my G28 command is configured to do a homing until X-min/Y-min/Z-min trigger.

Hywelmartin, 2019-03-09:
The homing direction is dependant on which max is closest to the endstop position
see the x stepper config in [example.cfg] (https://github.com/KevinOConnor/klipper/blob/9466c2d66cce7290bb73edcd7123d0b3a04a9021/config/example.cfg#L21)

HWiese1980, 2019-03-09:
So, hoping that I get my head around it any time soon... I cannot just place the two z-max endstops arbitrarily and to the Z0 homing by the probe? I have to provide an exact position for the z-max endstops? Because for only gantry tilt adjustment it's not necessary to know the height of the endstops as long as they are at the same hight. 

I'm not sure if that has come across correctly... I'm sorry if I repeat myself: I want to use the two z-max endstops **solely and exclusively** for z-gantry levelling/adjustment, **not** for z-homing (detecting where z currently is). The actual **homing** shall be done by the probe (inductive sensor).

Hywelmartin, 2019-03-09:
I'm trying to solve the problem with existing routines...

you could home to max on Z and then make 

```
G1 X100 Y100 Z20 ; go to the middle of build plate and down to Z 20
PROBE ; make single probe
G92 Z0 ; set position(should be your off set)
G1 Z10 ; go up a little
```
this alters the coordinate system(with the bed as the origin for Z) and makes the setup of the height for the endstops less critical ...

HWiese1980, 2019-03-09:
So I have to know relatively precisely where the endstops are in order to be able to do a G1 Z20 without risking a head crash, is that correct? Honestly, I'd rather rely on G28 to do the z-min homing with the probe...

Thus, my feature request remains: an independant, non-tinkering-with-basic-GCODE way to do what I'd like to do without touching the behavior of G28 as it currently is, i.e. in my case homing to z-min. 

If it doesn't exist like this already, as it seems, and if I wanted to implement it myself and make a PR, where would I start?

Hywelmartin, 2019-03-09:
The probe will trigger over a distance Z20 to position_min: -5 in the Z stepper settings, so within 2.5 cm in my example, you could command the the printer to other Z heights as well to gain a bigger possible error... I wouldn't consider that to be precisely..
I have a Delta bot and use this approach all the time to be able to switch nozzles, bed surface and even  heater blocks between normal and volcano...

A starting point  https://github.com/KevinOConnor/klipper/blob/master/docs/CONTRIBUTING.md
and have a look at klipper/klippy/extras

KevinOConnor, 2019-03-11:
I'm not sure there is anything further to add to this - the typical solution is to either home to the endstops, or to use z_tilt_adjust.  Since you don't want to home to the endstops, then you can use z_tilt_adjust.  (A number of people use z_tilt_adjust on "bed slingers" - the physical location of the XY carriages doesn't impact the calibration.)

-Kevin

HWiese1980, 2019-03-11:
Well, how does the z_tilt_adjust feature work if the vertical offset does not change in relation to X and Y? 

But anyway, I've set this up now using `homing_override`, homing to the endstops and, as @Hywelmartin suggested, `PROBE` and `G92`. It works. I don't really like it because it makes the probe's `z_offset` meaningless and abuses `G28`, but it works. 

I'd rather have it in a distinct command and have G28 do the unoverridden homing towards Z0 though, because I don't see the use in readjusting the gantry before every single print and I'm not sure how reliable a single `PROBE` is. Furthermore it takes time to move the gantry all the way up to adjust for tilt and all the way down again to do the `PROBE` based Z homing. I don't really want to do that every time a print starts. But since I can only override `G28` in its entirety including X and Y axis, I can only either do the gantry tilt adjustment by `homing_override` because it needs the homing mode (i.e. moving + waiting for the endstops to trigger). 

I still propose an additional command for this purpose that does what I'm doing now, i.e. **homing** to the max-Z endstops, but that leaves the original `G28` untouched.

Hywelmartin, 2019-03-11:
The probe reliability is probably good enough 
You could activate the force_move in config and make a home_macro like:

```
[gcode_macro HOME]
default_parameter_Z: 150
default_parameter_OFFSET: 2.3
gcode:
       G28 X Y
       SET_KINEMATIC_POSITION Z={Z}
       G1 X100 Y100
       PROBE
       SET_KINEMATIC_POSITION Z={OFFSET}
```

Then if you home it (G28) you will get "tilt" adjust
after this routine a bed level or bed tilt could improve the first layer

HWiese1980, 2019-03-11:
What's different here from what you suggested before, @Hywelmartin? Aside from `SET_KINEMATIC_POSITION` instead of `G92` that is

Hywelmartin, 2019-03-11:
You don't home the z-axis.. you tell the printer that it has a z position..

HWiese1980, 2019-03-12:
So I still `G28`-home the Z-axis towards max-Z/my two endstops, and then use `HOME` to sort of "home" it to Z0?

Hywelmartin, 2019-03-12:
No you Should only need to HOME if you are certain that the gantry is
level..
If the gantry need leveling you use G28 followed by a PROBE and G92 Zoffset
or the HOME macro

HWiese1980, 2019-03-12:
I see. Sorry if it takes some time for me to wrap my head around it. 

It appears to be already very close to what I originally proposed, aside from the fact that `G28` and `HOME` are swapped and thus `G28` equals the additional command that I proposed. In other words...

Now, as you suggest: 

* `HOME` to home Z to {probe:z_offset} iff I'm sure the gantry is level
* `G28` to level gantry

What I proposed:

* `LEVEL_GANTRY` (or something like that) to level the gantry by moving it towards the two endstops
* `G28` to home Z to {probe:z_offset} using the probe, as it is its normal behavior without all the gantry levelling stuff (I'd prefer to have it that way to keep consistency between `G28 X Y` and `G28 Z`)

KevinOConnor, 2019-03-14:
FWIW, if someone wishes to add a new levelling command and submit a pull request, then that is fine.  I imagine it will be difficult for a developer to do that without hands on access to the specific hardware though.

-Kevin

HWiese1980, 2019-03-14:
Yeah, that's why I'm trying to implement it myself. But I'm still trying to wrap my head around the code. No offense, rather some constructive criticism: it lacks docstrings. :wink:

KevinOConnor, 2019-03-31:
I'm going to close this issue as it looks like the conversation has concluded.  Feel free to submit a PR with an implementation.

-Kevin

closed 2019-03-31 by KevinOConnor

schnello, 2019-09-17:
@HWiese1980 could you solve your problem with the suggestion from @Hywelmartin 
'''
[gcode_macro HOME]
default_parameter_Z: 150
default_parameter_OFFSET: 2.3
gcode:
       G28 X Y
       SET_KINEMATIC_POSITION Z={Z}
       G1 X100 Y100
       PROBE
       SET_KINEMATIC_POSITION Z={OFFSET}
'''