Klipper3d/klipper #1377: Stepper too far in past with TMC2208 and LPC17xx

manoukianv, 2019-03-09:
Hi,

i have a issue when trying to do an home with tmc2208 setup @256 mstep on a Sbase (LPC17xx 100Mhz)
It's when printer will do the "G0 X113.5 Y169 F12000" in the homing_override section.

Is it mcu overflow ? Is there a way to setup this param ?

Best regards,
Vincent

[klippy.log](https://github.com/KevinOConnor/klipper/files/2948492/klippy.log)

BlackStump, 2019-03-09:
why 256 microsteps you gain nothing apart from overloading the mcu ...really try 16 microsteps with interpolation of 256
your step distance should be 
0,0125

shadrincev, 2019-03-10:
256 steps too many. The most optimal value is 128 for the x and y axes. For an extruder 64, this is if you have 2208 there.

KevinOConnor, 2019-03-10:
@manoukianv - I think you've answered your own question - the generated step rate is too high.

The reason it is too high is that the default step pulse time is 2us (which means it takes a minimum of 4us for each step pulse).  That mean, in practice, no single stepper motor can exceed about 200K steps per second.

If you are only using trinamic stepper drivers and you wish to run tests at high step speeds, then you can reduce the default step pulse time in "make menuconfig" (it's under the low-level options).

-Kevin

manoukianv, 2019-03-11:
Thanks, you can close the issue.
Why 256 : 
- because i want to test the effective gain in high mstep vs interpolate (measured)
- i have artefact at différent mstep which i don't have with drv8825
- i want to test more advanced parameters, and i don't read any limitation on mstep in docs, so i test...

FHeilmann, 2019-03-12:
@manoukianv you can close the issue too with the appropriate button at the bottom of this page.

closed 2019-03-15 by manoukianv