Klipper3d/klipper #1384: Something strange with a delta printer.

Nossolov, 2019-03-11:
Hello.
I  use Klipper for month now with my delta printer (very good !), and yesterday, suddenly, printed parts became too small (about 58mm for 60 needed). I first check all i can in mecanics without result. Several manual calibrations and no change : good print but too small.
Today, i delete everything relative too "save_config" action in config file, and redo a complete calibration, and now, it's great again. No idea what's append....

Nossolov, 2019-03-12:
Sorry...
just add.

[klippylog.txt](https://github.com/KevinOConnor/klipper/files/2958466/klippylog.txt)

KevinOConnor, 2019-03-14:
Okay, I'm not sure what happened either.  Best guess is that something in an "advanced delta calibration" caused the code to adjust the xy dimensions.  Should the problem occur again, grab the log from before resetting the config.

-Kevin

Nossolov, 2019-03-14:
I did just a normal manual calibration.  May be this comment help someone else.
Klipper is great !

closed 2019-03-14 by Nossolov