Klipper3d/klipper #1385: [Feature request] Merging STM32 targets

Delsian, 2019-03-12:
Actually not FR, but only question about better variant regarding PRs #1382 and #1383 
Is it have the sense to merge all STM32-related targets into single choice inside "Micro-controller Architecture" selection (like "STM32"), and move CMSIS and SRC folders one level deeper?
Probably it requires some significant changes in build system.
Or just leave this "as-is" for first step of new target integration and keep this in mind as "Future changes"?

KevinOConnor, 2019-03-14:
FWIW, I would do the latter - keep it in mind for later.  Lets merge what we have now, see if additional targets become available, and then decide if merging makes sense at that point.

-Kevin

Delsian, 2019-03-15:
Ok, anyway better to wait for merging thrid STM32 target. For two targets this proposal has no sense.
BTW, I'm thinking now on new SMM32G0 series. Unfortunately it not accessible yet in my location.
Thanks for reply.

KevinOConnor, 2019-03-22:
Okay, I'm going to close this as I think we've found consensus on the original question.

-Kevin

closed 2019-03-22 by KevinOConnor