Klipper3d/klipper #1386: Printing from "virtual SD" runs end-script multiple times. Scripts into FW-config possible?

swatti87, 2019-03-12:
I've inserted a simple fast retract and lift 10mm commands into my end-script and cancel-script. If i upload to "SD" via Octo adn try to print from there, and cancel the print, the script is run multiple times. Once my printer got stuck to a loop trying to re-start after it gave a wrong out-of-range error.

I've tried to add the scripts to both Octo and Cura, both behave the same.

Can the start & end scripts be added to the Klipper Config? I'd imagine it has the "final say" in all things happening.

jakep82, 2019-03-12:
The scripts you're referring to are run by Octoprint.  This isn't a Klipper issue.

swatti87, 2019-03-19:
Currently the scripts are running from Octo or Cura/slicing software. 

"Can i" type them to FW so it runs them from the Config-file? Meaning i can remove them from Octo and Cura.

closed 2019-03-20 by klipper-gitissuebot

swatti87, 2019-03-28:
Issue remains. 
Bots... Really...