Klipper3d/klipper #1390: MCU 'mcu' shutdown: No next step

Westra, 2019-03-13:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2962737/klippy.zip)
Hello all,

The shutdown occurred just after the start of the print, also had the same error at the end of some different prints. 

It happens sparely but is annoying of course :) The printer is a Voron 2.1 with 2x mks base controllers one for XYE and one for the four Z steppers.

Could you help me out?

Kind regards,
Westra

KevinOConnor, 2019-03-14:
This appears to be a defect in Klipper.  I will need to investigate further.

-Kevin

Westra, 2019-03-19:
I'm always happy to help by finding defects :) - if you need more info I will try to provide it.

Kind regards,
Westra

Westra, 2019-03-21:
same error different print, attached log.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2991675/klippy.zip)

KevinOConnor, 2019-03-30:
I believe I've tracked this down - it's an issue with the minimum velocity safety check itself.  Should be fixed now (commit 75b73b09).

-Kevin

Westra, 2019-04-01:
Nice!
I will update klipper, and close the issue
Tnx, Westra

closed 2019-04-01 by Westra