Klipper3d/klipper #1393: [Feature request] ATX auto power on/off with trigger gcode

tripstar76, 2019-03-14:
Would it be possible to impliment an auto power on using trigger gcodes like the octoprint plugin PSU Control uses.
They use "G0,G1,G2,G3,G10,G11,G28,G29,G32,M104,M106,M109,M140,M190" as trigger codes to switch a pin.
I currently use a macro to turn on for print starts. But having it trigger when klipper reads one of the above commands would be ideal.
If not, would there be a way that this could be done through the config file?

Hywelmartin, 2019-03-15:
you could try https://github.com/KevinOConnor/klipper/blob/378402b4d68c41fb74c02164d10b9161f65a1b4d/config/example-extras.cfg#L568

tripstar76, 2019-03-15:
Ahh right. So, I have my PSU Power-ON wire connected to pin12 and this works...

`# set pin for PSU`
`[output_pin ATX]`
`pin: ar12`
`value: 1`
`shutdown_value: 1'`
` `
`# power on`
`[gcode_macro M80]`
`gcode: SET_PIN PIN=ATX VALUE=0`
` `
`# power off`
`[gcode_macro M81]`
`gcode: SET_PIN PIN=ATX VALUE=1`

So, I could get rid of the above code, and do...

`[controller_fan my_PSU_ON]`
`pin: ar12`
`max_power: 1.0`
`shutdown_speed: 0 # turn PSU off if error?`
`#cycle_time:`
`hardware_pwm: True # overide the above cycle time so the pin is pmw?`
`kick_start_time: 0.0`
`idle_timeout: 600 # turn PSU off after 10mins?`
`#idle_speed: # leave as default max_power?`
`heater: extruder, heater_bed, stepper_x, stepper_y, stepper_z`

And then get rid of any instances of M80/M81?

tripstar76, 2019-03-15:
ok, this didn't work, until I removed `stepper_x, stepper_y, stepper_z`.
Although it says `enabled whenever its associated heater or any configured stepper driver is active.`
and `If a comma separated list of heater names is provided here, then the fan will be enabled when`
`any of the given  heaters are enabled.`
So, can this be used on steppers that are called?
I'm a little tired and a bit confused, so please forgive my ignorance

jakep82, 2019-03-15:
Steppers are not heaters.  Any configured stepper means **_any_** stepper in your config.  You don't need to specify them in the controller_fan section.

tripstar76, 2019-03-16:
Thanks, I'd misunderstood it.
It works now

closed 2019-03-16 by tripstar76