Klipper3d/klipper #1395: After 4 months of not printing i cant connect to printer.

Alennartsson, 2019-03-14:
So basically i finally got my printer home again and i was so excited to start printing. The problem is that i cant connect to my printer via octoprint. This was never an issue before, i did allways leave the raspberry on and restarted my printer when i wanted to print. 

But now i just get this in the terminal when i try to connect, i have no single idea on what to try anymore. Have been sitting with this problem for 5 hours now.

Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x69c39e10, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor

theopensourcerer, 2019-03-14:
Is klipper running?
Has the USB port changed with respect to what is in your printer.cfg file?

Suggest taking a look through the installation documentation and faq etc...

Alennartsson, 2019-03-14:
> Is klipper running?
> Has the USB port changed with respect to what is in your printer.cfg file?
> 
> Suggest taking a look through the installation documentation and faq etc...

[stepper_x]
step_pin: PD7
dir_pin: !PC5
enable_pin: !PD6
step_distance: .01
endstop_pin: ^!PC2
position_endstop: -14
position_max: 220
position_min: -14
homing_speed: 80

[stepper_y]
step_pin: PC6
dir_pin: PC7
enable_pin: !PD6
step_distance: .01
endstop_pin: ^!PC3
position_endstop: -1
position_min: -1
position_max: 220
homing_speed: 80

[stepper_z]
step_pin: PB3
dir_pin: !PB2
enable_pin: !PA5
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#endstop_pin: ^!PC4
#position_endstop: 1.78
position_min: -10
position_max: 240
homing_speed: 20

[virtual_sdcard]
path: ~/.octoprint/uploads/

[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
step_distance: .006644547
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA7
control: pid
pid_Kp: 19.190
pid_Ki: 0.615
pid_Kd: 149.679
min_temp: 0
max_temp: 250

[verify_heater extruder]
heating_gain: 1
check_gain_time:60
hysteresis: 30
max_error: 130

[heater_bed]
heater_pin: PD4
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA6
control: pid
pid_Kp=74.611 
pid_Ki=1.675 
pid_Kd=830.979
min_temp: 0
max_temp: 130

[fan]
pin: PB4

[mcu]
#serial: /dev/ttyUSB0
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
baud: 250000

[printer]
kinematics: cartesian
max_velocity: 400
max_accel: 1500
max_z_velocity: 20
max_z_accel: 100

[gcode_macro M900]
gcode: SET_PRESSURE_ADVANCE ADVANCE={K}

[display]
lcd_type: hd44780
rs_pin: PA3
e_pin: PA2
d4_pin: PD2
d5_pin: PD3
d6_pin: PC0
d7_pin: PC1

# Define the BLTouch servo
[servo bltouch]
pin: PA4
maximum_servo_angle: 180
minimum_pulse_width: 0.0006 
maximum_pulse_width: 0.0024
# SET_SERVO SERVO=bltouch ANGLE=x
# 10 = push pin down
# 60 = endstop test mode
# 90 = pull pin up 
# 120 = self test (pin up & down)
# 160 = release alarm / reset

#bltouch
[probe] 
pin: ^PC4 
x_offset: 29  
y_offset: 0
z_offset: 1.91
  
activate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=10
    SET_SERVO SERVO=bltouch ANGLE=60
    G4 P200
    SET_SERVO SERVO=bltouch ANGLE=10
    SET_SERVO SERVO=bltouch ANGLE=60
    G4 P200 
deactivate_gcode:
    SET_SERVO SERVO=bltouch ANGLE=90
    G4 P100 

[bed_tilt]
#x_adjust:
#y_adjust:
#z_offset: 2.50

points:
    84,110
    -14,10
    -14,110
    -14,210
    81,10
    81,210
    176,10
    176,110
    176,210

[homing_override] 
set_position_z: 5 
axes: z 
gcode:
    G90 ; Uncomment these 2 lines to blindly lift the Z 2mm at start
    G1 Z7 F600
    G28 X0 Y0
    #G1 X104 Y110 F3600
    G1 X110 Y110 F3600
    SET_SERVO SERVO=bltouch ANGLE=160 ; reset
    SET_SERVO SERVO=bltouch ANGLE=90 ; pin up
    G4 P1000 ; wait 1 s
    G28 Z0
    SET_SERVO SERVO=bltouch ANGLE=160 ; reset
    G4 P1000 ; wait 1 s
    SET_SERVO SERVO=bltouch ANGLE=90 ; pin up

jakep82, 2019-03-14:
I'm sure the bot will be here shortly, but you'll need to post a log file to get any help.

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#issue-reporting

Alennartsson, 2019-03-14:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2968531/klippy.log)

Alennartsson, 2019-03-14:
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2968542/printer.cfg.txt)

Alennartsson, 2019-03-15:
No1 that can help? I do really want to print and im stuck with this problem

theopensourcerer, 2019-03-15:
If you look in your klippy.log there is nothing after the initial start up messages - there should be *LOTS* of data in this file. For example there should be a dump of the printer.cfg file immediately after the line which reads:

```Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'```

So it looks like klipper tries to start a few times and then stops.

I would guess you will need to do some debugging on the Raspberry Pi using command line tools, e.g. try starting klipper from the command line and see if you get any errors... 

Unfortunately I think there is very little for anyone to go on to help you, as all we can see is klipper does not start.

If you do not have these skills, it might be a better idea to simply copy your printer.cfg file to somewhere safe and the follow the installations instructions by starting from the very beginning again, i.e. from a clean SD card.

https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

theopensourcerer, 2019-03-15:
One minor thing in your printer.cfg file... I have not seen this syntax before (with the {}). It might be OK but it might be worth removing that section and trying to restart...

```
[gcode_macro M900]
gcode: SET_PRESSURE_ADVANCE ADVANCE={K}
```

Alennartsson, 2019-03-15:
> If you look in your klippy.log there is nothing after the initial start up messages - there should be _LOTS_ of data in this file. For example there should be a dump of the printer.cfg file [...]

I do also think that its something wrong on my raspberry and not the firmware itself. So you basically think its better to just reinstall klipper on the reaspberry?

I mean it should be fixable because it was working 4 months ago and i have not made any changes.

Alennartsson, 2019-03-15:
And just tell me what more information you guys need and how i get it and i will fix it.

Alennartsson, 2019-03-15:
[image]

Alennartsson, 2019-03-17:
[image]

I was trying to update klipper and followed the guide. After doing so i got this problem when trying to run the make command? 

Any help?

Alennartsson, 2019-03-17:
Solved everything with a clean install on a new sd card. Strange that issues like this just can come up without using it.

closed 2019-03-17 by Alennartsson