Klipper3d/klipper #1396: What causes this?

JustAnother1, 2019-03-14:
I print using Klipper on a delta printer. In my print I see these curved offsets:

[image]

I think you can see the Z Layer lines. The curved disturbance goes over many layers. It is also on all sides. I have never seen something like this. Due to the curved nature it could be something related to the delta movement. Does anybody know if there is something wrong on the printer that causes this? I checked all things I can think of and found nothing, but I might have missed the critical thing.

Or could this be a "mathematical problem" in how Klipper calculates the moves for deltas? Rounding issues or something?

Does anybody know this effect or also has is on a delta printer? Or has a delta printer running Klipper and does not have this effect?

BlackStump, 2019-03-15:
I use Klipper with my Delta printer and do not have that sort of artifact in my prints so that would suggest it is not a klipper problem but with so many variables and no details of your Delta printer or how you have configured it and no log file it would be any ones guess.

Hywelmartin, 2019-03-15:
some fault i the motion system on the axis that is corresponding to that sphere (radius) so to the left according to the picture

So this is not a Klipper issue...

JustAnother1, 2019-03-16:
@Hywelmartin was right. I found a zip tie that interfered with the up and down movement on one of the towers. By fixing this the issue disappears.

Thank you everyone for your help.

closed 2019-03-16 by JustAnother1