Klipper3d/klipper #1397: Spurious probe trigger during Y axis movement during BED_MESH_CALIBRATE

fcollingwood, 2019-03-15:
I'm using underbed piezo sensors for both Z endstop and Z probe. What I'm noticing is that the probe is triggering during Y moves (hasn't happened in X yet), and Klipper is counting this as a probe activation when the bed comes to a stop, which completely throws things off.

I've only just installed Klipper and am in the process of getting it up and running. Previously I was using Marlin, and it was not an issue as Marlin ignores any probe triggers during X and Y axis movement, it only uses probe triggers that happen during Z movement.

There's also another enhancement I'd like to request, again related to how Marlin probes. Marlin has a setting where the user can define a time setting between reaching a probing position, and starting the probe - eg. bed moves in Y axis to probing position, system pauses for n seconds to allow system to settle, then does Z probe.
[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/2970661/klippy.log.gz)

theopensourcerer, 2019-03-15:
The pause is pretty easy. See the `activate_gcode:` option in https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg. 

e.g. Add a 
```G4 P750```
This will pause for 750 mS between movement to the probe point and actually probing.

fcollingwood, 2019-03-15:
That seems to have fixed it. Thank you!

theopensourcerer, 2019-03-15:
Awesome. I use a Precision Piezo probe and it is so sensitive I can't walk around the room when it is probing ;-)

Am not sue if you are able to but if there is a button, can you close this ticket?

fcollingwood, 2019-03-15:
I use the Precision Piezo board, but I designed my own underbed piezo system as I wanted a really simple carriage. I use the same username on here and Thingiverse, you can find it there.

I see the close button, so will do

closed 2019-03-15 by fcollingwood