Klipper3d/klipper #1399: Delta Kinematics issue

casedog21, 2019-03-15:
New to klipper and I tried to search for an answer before I posted.
In my config when I fire off a G28 it homes or when I use a move command.

In my config it has it set to delta but when I home it only moves one stepper at a time. 
If I send a g0 z450 as an example, it is only moving one stepper instead of all three. Not sure ihat I am doing wrong here. Once I get this worked out, I will share my config. I have a Tevo Little Monster with a mks sbase 1.3 board.

Thanks for any help in advance.
-Casey

`# This file contains common pin mappings for Smoothieboard. To use
# this config, the firmware should be compiled for the LPC176x.

# See the example.cfg file for a description of available parameters.
# LEFT
[stepper_a]
step_pin: P2.0
dir_pin: !P0.5
enable_pin: !P0.4
step_distance: .0125
#endstop_pin: ^P1.24
endstop_pin: P1.25
position_endstop: 510
#position_max: 240
homing_speed: 50
arm_length: 397.1900

#back
[stepper_b]
step_pin: P2.1
dir_pin: !P0.11
enable_pin: !P0.10
step_distance: .0125
#endstop_pin: ^P1.26
endstop_pin: P1.27
position_endstop: 0
#position_max: 240
homing_speed: 50
arm_length: 397.1900

#right
[stepper_c]
step_pin: P2.2
dir_pin: !P0.20
enable_pin: !P0.19
step_distance: .0025
#endstop_pin: ^P1.28
endstop_pin: P1.29
position_endstop: 0.5
#position_max: 500
arm_length: 397.1900

[extruder]
step_pin: P2.3
dir_pin: P0.22
enable_pin: !P0.21
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.23
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250

#[extruder1]
#step_pin: P2.8
#dir_pin: P2.13
#enable_pin: !P4.29
#heater_pin: P2.6
#sensor_pin: P0.25
#...

[heater_bed]
heater_pin: P2.5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.24
control: watermark
min_temp: 0
max_temp: 130

[fan]
pin: P2.4

[mcu]
serial: /dev/ttyACM0
#baud: 115200

[printer]
#name: TevoLittleMonster
kinematics: delta
max_velocity: 500
max_accel: 3000
max_z_velocity: 200
#max_z_accel: 100
delta_radius: 120

[static_digital_output leds]
pins: P1.18, P1.19, P1.20, P1.21, P4.28

[mcp4451 stepper_digipot1]
i2c_address: 44
# Scale the config so that wiper values can be specified in amps.
scale: 2.25
# wiper 0 is X (aka alpha), 1 is Y, 2 is Z, 3 is E0
wiper_0: 1.0
wiper_1: 1.0
wiper_2: 1.0
wiper_3: 1.0

[mcp4451 stepper_digipot2]
i2c_address: 45
scale: 2.25
# wiper 0 is E1
wiper_0: 1.0

# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
#[display]
#lcd_type: st7920
#cs_pin: P0.16
#sclk_pin: P0.15
#sid_pin: P0.18
#encoder_pins: ^P3.25, ^P3.26
#click_pin: ^!P1.30
`

casedog21, 2019-03-15:
Here is my log

[klipy.log](https://github.com/KevinOConnor/klipper/files/2972438/klipy.log)

jakep82, 2019-03-15:
You endstop positions don't look right.  I'm not terribly familiar with deltas, but why is stepper_a 510, stepper_b 0, and stepper_c 0.5?

casedog21, 2019-03-15:
good catch but I dont think that will effect my problem. Will try it and see

casedog21, 2019-03-15:
same issue

casedog21, 2019-03-15:
M114 looks right now
but it still seems to be treating it like a cart vs a delta

jakep82, 2019-03-15:
You also have an issue with step distances.  Stepper_c is wrong

casedog21, 2019-03-15:
`Send: M114
Recv: X:0.000 Y:-0.000 Z:450.000 E:0.000
Recv: ok`

Instead of the entire gantry dropping down to 450 it only moved one leg.

casedog21, 2019-03-15:
That fixed it!  Thank you!!!

closed 2019-03-15 by casedog21

wiwatto, 2019-03-17:
"casedog21"  please post you  klippy.log  files after  "FIXED"